bvanheu/stratatools #15: bp_read.py error

RobListon, 2015-05-06:
Hi,
I'm trying to use a bus pirate v3 to read one of my material cartridge chips and I am getting this error:
Traceback (most recent call last):
  File "./bp_read.py", line 135, in <module>
    print("Device found: " + rom_sequence)
TypeError: cannot concatenate 'str' and 'NoneType' objects

Am I missing something obvious here to fix it?

Bald888Eagle, 2015-05-06:
What version of Python are you running?
I'm just guessing but the following may fix it.
print("Device found: " + str(rom_sequence))

RobListon, 2015-05-06:
I'm got python 2.7.3 installed right now.
Making the edit you suggested fixed that problem and now I've run into a new one:
Device found: None
Traceback (most recent call last):
  File "./bp_read.py", line 138, in <module>
    raise(Exception("unable to find a device on this 1-wire bus"))
Exception: unable to find a device on this 1-wire bus

RobListon, 2015-05-28:
Upgrading to python 2.7.9 fixed the issue. Anything above 2.7.5 should work as well.

closed 2015-05-28 by RobListon