bvanheu/stratatools #16: Stratasys Fortus

rowlandski, 2015-05-07:
Hi,

will your code work on Fortus cartridges?

And which version Raspberry Pi did you use, or doesn't if matter? 

Many thanks.

Regards,

Rowland

bvanheu, 2015-05-07:
It should work with a Fortus 250.

I used a Raspberry Pi 1, but I'm not sure regarding the RP2. The code will work, but the 1wire might be on another pin.

rowlandski, 2015-05-14:
Thanks for your reply.

I will be using/editing a Fortus cartridge.

What should i use for machine "type"?

Print information about a cartridge

You have to provide the machine type (fox, prodigy or quantum) and the EEPROM uid, in hexadecimal form without the '0x' prefix. Note that the EEPROM uid use to end with "23".

$ ./stratasys-cli.py eeprom -t fox -e 4343434343434343 -i random_file.bin

Many thanks.

rowlandski, 2015-05-28:
Prodigy works for Fortus (250).

closed 2015-05-28 by rowlandski