bvanheu/stratatools #17: w1-wire

rowlandski, 2015-05-12:
Hi,

I'm trying to find and load:
$ sudo modprobe w1-wire w1-ds2433

There's a w1_ds2433, which I'm hoping is the same as I believe modprobe will use "-" or "_".

But, I can not find: w1-wire, will w1-gpio do the same? Or do I need to download w1-wire from somewhere?

Any comments or suggestions welcome.

Many thanks.

Regards,

Rowland

RobListon, 2015-05-12:
I'm also having this issue when trying it using my RPI. I'm pretty sure it's because I have a RPI v2 and w1-wire needs to be replaced with w1-gpio instead.

rowlandski, 2015-05-12:
Yes, I think you're right as I have a RPi ver2 too.

I've just created a script to load w1_gpio and w1_ds2433

I'm now going on to the next part of the read.me.....

rowlandski, 2015-05-12:
Yes, I will be doing that... thanks.

I'm just trying:
"You should now see your eeprom appearing:
$ ls -l /sys/bus/w1/devices/w1_bus_master1
23-xxxxxxxxxxxx/
(...)"

But get back, "No such file or directory"... and there is no file!

You got it working?

RobListon, 2015-05-12:
Yes,
When I run "ls /sys/bus/w1/devices" I see 23-0000014b4065 and w1_bus_master1.
running "ls /sys/bus/w1/devices/23-0000014b4065" get me this:
driver  eeprom  id  name  power  subsystem  uevent

rowlandski, 2015-05-12:
Are you connected as per the wiring diagram? No changes?

Just load the w1_gpio & w1_ds2433
then "ls /sys/bus/w1/devices"... it then works?

RobListon, 2015-05-12:
Yes, I've got it wired per the diagram.
You may need to edit your /boot/config.txt file as well and then reboot.
I added this line:
dtoverlay=w1-gpio,gpiopin=4

bvanheu, 2015-05-12:
Rob Liston have good point, you might have to specify the gpio poin used! Although I think it doesn't matter if the 1wire is configured by the device tree or probed by the user in the kernel.

Running the following should be enough:
> modprobe w1-gpio gpiopin=4

rowlandski, 2015-05-13:
Thanks for your help.

I've tried:
modprobe w1-gpio gpiopin=4
But I don't get any comms out of pin 4 (I have a scope on it)

As for editing /boot/config.txt and adding "dtoverlay=w1-gpio,gpiopin=4"
I don't have a config.txt file (I'm new to RPi's, only had one for a couple of days).

Is it a case of creating one?

Thanks.

RobListon, 2015-05-13:
Yes, you can create your own config.txt file. The documentation for it is available here: http://www.raspberrypi.org/documentation/configuration/config-txt.md

rowlandski, 2015-05-14:
I've found mine by holding down shift while booting.

I actually followed the instructions a few days ago regarding the config.txt, it seems that the tech note is not 100% correct. 

But in any case, i've got the EEPROM's ID but now.

Thanks for your reply.

rowlandski, 2015-05-14:
Also FYI:

It's best to use dtoverlay=w1-gpio,gpiopin=4, and place it in the config.txt.

Modules and now been load by dtoverlay in the config.txt file. (they are moving away from using the /modules.. modprobe).

HTH

closed 2015-05-28 by rowlandski