bvanheu/stratatools #18: No module named cartridge

RobListon, 2015-05-12:
I'm trying to run stratasys-cli.py to read the info on a cartridge I dumped but I'm running into a problem.
The command I'm running is "python3 ./stratasys-cli.py eeprom -t prodigy -e b60000014b406523 -i ~/eeprom-2365404b010000b6.bin" and this is the error:
Traceback (most recent call last):
  File "./stratasys-cli.py", line 36, in <module>
    from stratasys import manager
  File "/home/pi/stratasys/stratasys/manager.py", line 29, in <module>
    import cartridge
ImportError: No module named cartridge

rowlandski, 2015-05-14:
Rob,

I'm just about to do the same....

With regards to "cartridge", I have the DS2433 EEPROM board from the cartridge and have wired directly. I have never seen a cartridge, and how it connects. I'm assuming there shouldn't be any issues?

I have noticed your post says Python 3, I did read somewhere on this forum that the code only supports Python 2 (from my understanding).

I have a quick question, regarding the pycrypto, where does it need to live?

Regards,

RobListon, 2015-05-14:
I installed Raspian via the NOOBS installer so I only had python 2.7.3 which caused a bunch of problems. I downloaded and installed python 2.7.9 from the source package to resolve those issues.
As for pycrypto, I just downloaded that from https://www.dlitz.net/software/pycrypto/ then unpacked it and ran "pyhton setup.py build" and "python setup.py.install" to install it.

rowlandski, 2015-05-15:
Hi, thanks for the info.
I too used NOOBS installer and installed Raspian. I'll check what version of Python i'm running.
As for pycrypto, i did the same: build / install, but got errors.

I thought all this was going to be straight forward, but it's proving to be very character building!

I've put a post on pycrypto to see if i get any info back.

rowlandski, 2015-05-15:
pi@raspberrypi ~ $ python --version
Python 2.7.3

Think I need to upgrade.

RobListon, 2015-05-15:
After installing Python 2.7.9 all of the strange messages I was getting went away. I also tested using 2.7.6 on another computer. As far as I can tell, as long as you have 2.7.6 or higher you should be fine.
Once you upgrade you'll probably need to install pycrypto again as well.

rowlandski, 2015-05-18:
Rob,

I have python 2.7.9 and pycrypto now installed.

I'm now running the stratasys-cli command and getting an error, did you resolve your problem?

 pi@raspberrypi ~/Downloads/stratasys-master $ sudo python stratasys-cli.py eeprom -t fox -e 0000016c560823 -i random_file.bin
Traceback (most recent call last):
  File "stratasys-cli.py", line 228, in <module>
    app.run()
  File "stratasys-cli.py", line 47, in run
    args.func(args)
  File "stratasys-cli.py", line 125, in command_eeprom
    self._eeprom_info(args)
  File "stratasys-cli.py", line 167, in _eeprom_info
    cartridge = m.decode(machine_number, args.eeprom_uid, cartridge_crypted)
  File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 73, in decode
    cartridge_packed = self.decrypt(machine_number, eeprom_uid, cartridge_crypted)
  File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 213, in decrypt
    key = self.build_key(cartridge_crypted[0x48:0x50], machine_number, eeprom_uid)
  File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 235, in build_key
    key[0] = ~cartridge_key[0] & 0xff
IndexError: bytearray index out of range
pi@raspberrypi ~/Downloads/stratasys-master $ 

index out of range.... :(

bvanheu, 2015-05-28:
Duplicate from the other issue you filled?

closed 2016-12-15 by RobListon

RobListon, 2016-12-15:
My problem was solved by not using Python3 and making sure I had python at version 2.7.6 or higher. Also, I needed python-dev installed to build pycrypto.