bvanheu/stratatools #19: eeprom uid transposer

RobListon, 2015-05-14:
Is it possible to add option for a transposer or formatter function to this project so that I can enter the UID of a chip as it is displayed at /sys/bus/w1/devices/ (eg. 23-0000014b4065) and have it converted to the UID (eg. b60000014b406523) used to encrypt/decrypt the eeprom?
It's not a big deal but would help prevent an accidental typo from messing things up.
Thanks.

rowlandski, 2015-05-19:
Rob,
Transposing. The example given:
233a38b1020000c0", you should reverse it to be "c0000002b1383a23".
ie reversing the bytes.

It doesn't look like you've done that, but only moved "23".

And where did your "b6" come from?

Regards,

Rowland

bvanheu, 2015-05-28:
It is possible ;)

I'm currently in the process of making sure the software work in Python3, then I'll move into making it easier to use.