bvanheu/stratatools #20: Array oout of range error....

rowlandski, 2015-05-18:
I have python 2.7.9 and pycrypto now installed.

I have a Fortus 250, not sure which cartridge "type" to use? Fox / Prodigy?

I'm now running the stratasys-cli command and getting an error, did you resolve your problem?

pi@raspberrypi ~/Downloads/stratasys-master $ sudo python stratasys-cli.py eeprom -t fox -e 0000016c560823 -i random_file.bin
Traceback (most recent call last):
File "stratasys-cli.py", line 228, in
app.run()
File "stratasys-cli.py", line 47, in run
args.func(args)
File "stratasys-cli.py", line 125, in command_eeprom
self._eeprom_info(args)
File "stratasys-cli.py", line 167, in _eeprom_info
cartridge = m.decode(machine_number, args.eeprom_uid, cartridge_crypted)
File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 73, in decode
cartridge_packed = self.decrypt(machine_number, eeprom_uid, cartridge_crypted)
File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 213, in decrypt
key = self.build_key(cartridge_crypted[0x48:0x50], machine_number, eeprom_uid)
File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 235, in build_key
key[0] = ~cartridge_key[0] & 0xff
IndexError: bytearray index out of range
pi@raspberrypi ~/Downloads/stratasys-master $

index out of range.... :(

rowlandski, 2015-05-19:
Apologies, I hadn't entered the correct UID. So, have tried again and get this:

pi@raspberrypi ~/Downloads/stratasys-master $ xxd -p /sys/bus/w1/devices/w1_bus_master1/23-0000016c5608/id2308566c01000029
pi@raspberrypi ~/Downloads/stratasys-master $ sudo python stratasys-cli.py eeprom -t fox -e 290000016c560823 -i random_file.bin
Traceback (most recent call last):
  File "stratasys-cli.py", line 228, in <module>
    app.run()
  File "stratasys-cli.py", line 47, in run
    args.func(args)
  File "stratasys-cli.py", line 125, in command_eeprom
    self._eeprom_info(args)
  File "stratasys-cli.py", line 167, in _eeprom_info
    cartridge = m.decode(machine_number, args.eeprom_uid, cartridge_crypted)
  File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 73, in decode
    cartridge_packed = self.decrypt(machine_number, eeprom_uid, cartridge_crypted)
  File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 213, in decrypt
    key = self.build_key(cartridge_crypted[0x48:0x50], machine_number, eeprom_uid)
  File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 235, in build_key
    key[0] = ~cartridge_key[0] & 0xff
IndexError: bytearray index out of range
pi@raspberrypi ~/Downloads/stratasys-master $ 

Still have out of range.

RobListon, 2015-05-20:
I think these commands will work for you:
ls -l /sys/bus/w1/devices/
xxd -p /sys/bus/w1/devices/23-0000016c5608/id
sudo python stratasys-cli.py eeprom -t fox -e 290000016c560823 -i random_file.bin

rowlandski, 2015-05-21:
I get the same ID back, rotate it as suggested:
2308566c01000029 to
290000016c560823

And use:
pi@raspberrypi ~/Downloads/stratasys-master $ sudo python stratasys-cli.py eeprom -t fox -e 290000016c560823 -i random_file.bin

And still get an error:
...
File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 235, in build_key
key[0] = ~cartridge_key[0] & 0xff
IndexError: bytearray index out of range

I have a Fortus cartridge, and in teh notes it suggests to use "-t ktype", but that throughs an error straight back indicating must use; fox, fox2, uprint, etc.... 

Not sure what to do now.

Rowland

RobListon, 2015-05-21:
My machine is a Dimension 768 (a similar but much older model to the Fortus) and I'm using the prodigy type for it. Perhaps yours uses the same?

rowlandski, 2015-05-21:
Thanks for your reply, I have just tried "pordigy" and I get the same error message again (as above - "bytearray index out of range")

Have you manage to read and reprogram your eeproms/cartridges?

RobListon, 2015-05-21:
Yes, I just wrote a new file to one of my cartridges yesterday and printed with it.
Have you tried the other machine types like the fox 2 yet?

rowlandski, 2015-05-21:
No, I only have some cards from a Fortus 250 machine.

RobListon, 2015-05-21:
Sorry, I meant have you tried using other machine types in the stratasys-cli.py command?
Eg. "sudo python stratasys-cli.py eeprom -t fox2 -e 290000016c560823 -i random_file.bin"

rowlandski, 2015-05-22:
Yes, I'll tried all the options; fox, fox2, ktype, prodigy, quantum, uprint & uprintse.
ktype - reports no such type
All the rest give the bytearray error as indicated above.

Not sure what to do next....

RobListon, 2015-05-26:
I'm just guessing here but perhaps the file is corrupt? can you run "xxd -p random_file.bin" and paste the results here?
Either that or maybe your version of python is the cause. I ran into the same problems with python 2.7.3 and other issues with python 3. It wasn't until I compiled python 2.7.9 directly on my RPI that it worked like it was supposed to.

rowlandski, 2015-05-27:
I've had a look at the file, random_file.bin, and it's empty.
I've checked to see what version of python I'm running and it's 2.7.9.
I've just looked on https://www.python.org/downloads/ and Python version 2.7.10 is now available. 

Maybe worth installing 2.7.10 and see if that helps / fixes the issue?

rowlandski, 2015-05-27:
pi@raspberrypi ~/Downloads/stratasys-master $ python --version
Python 2.7.10
pi@raspberrypi ~/Downloads/stratasys-master $ sudo python stratasys-cli.py eeprom -t prodigy -e 2d0000017c4e9223 -i random_file.bin
Traceback (most recent call last):
  File "stratasys-cli.py", line 228, in <module>
    app.run()
  File "stratasys-cli.py", line 47, in run
    args.func(args)
  File "stratasys-cli.py", line 125, in command_eeprom
    self._eeprom_info(args)
  File "stratasys-cli.py", line 167, in _eeprom_info
    cartridge = m.decode(machine_number, args.eeprom_uid, cartridge_crypted)
  File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 73, in decode
    cartridge_packed = self.decrypt(machine_number, eeprom_uid, cartridge_crypted)
  File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 213, in decrypt
    key = self.build_key(cartridge_crypted[0x48:0x50], machine_number, eeprom_uid)
  File "/home/pi/Downloads/stratasys-master/stratasys/manager.py", line 235, in build_key
    key[0] = ~cartridge_key[0] & 0xff
IndexError: bytearray index out of range                     <----------------
pi@raspberrypi ~/Downloads/stratasys-master $ 

Get the same error message, array out of range.
:(

rowlandski, 2015-05-27:
I've just been reviewing the documentation and 
Dependencies
    python 2.7
    pycrypto

Have you tried using version 2.7 (2.7.0)?

bvanheu, 2015-05-27:
You need at least Python 2.7.5 (there is a bug in the earlier version regarding the struct and bytearray handling).

If `random_file.bin` is empty, then it won't work, you won't be able to decrypt it.

rowlandski, 2015-05-27:
bvanheu, thank you for your reply.

Regarding the random_file.bin file. I didn't have one so I created one. Was that incorrect?

Regards,

Rowland

rowlandski, 2015-05-27:
Sorry, I've worked it out. 

Sorry for stupid questions.

bvanheu, 2015-05-27:
> Regarding the random_file.bin file. I didn't have one so I created one. Was that incorrect?

Run this first:
cp /sys/bus/w1/devices/w1_bus_master1/23-xxxxxxxxxxxx/eeprom ~/eeprom.bin

Then replace `random_file.bin` by `~/eeprom.bin`

> Sorry, I've worked it out.

;)

rowlandski, 2015-05-28:
Thanks for your assistance.

Yes, you can close this ticket.

Regards.

closed 2015-05-28 by rowlandski

RobListon, 2015-05-28:
Hi Rowland,
Can I ask if you've been able to write the bin file back to the Fortis material cartridge and use it? I'm thinking of maybe purchasing this model of 3d printer and wanted to make sure I could do the refills myself.