bvanheu/stratatools #21: UPrint cartridge.

mederux, 2015-05-20:
Good evening,

I checked the wiring suggested on the readme, however I have an UPrint and its EEPROM is different as you can see on the images bellow, it has 4 contact areas (even if I believe it uses only 2) do you have any ideas on which ones should I use?
As recently you included the uprint machines ID's I do believe that someone already used it, if someone could help me in how to use it for this specific printer?

Thank you very much in advance!
[image]
[image]

milesguidon, 2015-05-21:
Hi Mederux, 

I have a uPrint SE Plus. You are correct that to access the EEPROM data you need only two data lines. I solder directly to the pads and go through a converter to my PC USB port. It's tough to describe which pads to use unless you can reassemble the plastic dongle. Do that, then take a look at this photo, which is for a model material dongle; take note of the orientation to know which pads to use. The unconnected wire is for data, and the connected one is going to a green wire, and that's the ground wire. Good luck!

It's good you have a uPrint and know the machine code! I have a uPrint SE Plus and do not yet know the machine code. I am assuming that the uPrint SE has a different machine code from the uPrint SE Plus, because I have tried doing my EEPROMS with the uPrint SE code, and it doesn't work. I flash all of my other dimension printer eeproms, so I'm familiar with the process.
\
[image]

mederux, 2015-05-25:
Hi milesguidon,

Thank you very much!
I tried using an Arduino, however i was using a wrong resistor (2.2KOhms instead of 4.7KOhms) I will replace the resistor and try again.

Were you using a raspberry Pi or the Bus Pirate?

Cheers!

milesguidon, 2015-05-25:
I am using a USB to RJ45 or RJ11 converter, with two wires stripped out of the RJxx jack to go to the cartridge with alligator clips. 

No need for a resistor or external equipment in my setup. It's pretty simple: I just use a PC with Maxim's OneWireViewer software to get the serial number and 55aa55xxxxxxxxxx codes, then just copy and paste the text of the python input into a command line window. Then, I use "bin2hex.exe" to convert the *.bin file that the python code generates into a hex file, then I run a keyboard macro in Zeus text editor to change the intelhex format to the format displayed by the OneWireViewer software. Once the keyboard macro is done, I just copy and paste the several lines of hex from Zeus to OneWireViewer software and I'm done. 

The whole process takes 2 minutes, tops, and you just need a PC.

dschiedsch, 2015-05-26:
Hi
Has anyone of you managed to write back to the EEPROM of the uprint printer?
As far as I know the write procedure of these new EEPROMs is password
protected in hardware (new EEPROM chip with password)

bvanheu, 2015-05-27:
> As far as I know the write procedure of these new EEPROMs is password protected in hardware (new EEPROM chip with password)

Exactly, for now you cannot write over a UPrint EEPROM for that specific reason :( I will add a note in the README.

SamuelBrucksch, 2015-10-14:
Hi,

i face the same problem right now. I can read but not write back. is there any workaround known yet?

Cheers

Samuel

bvanheu, 2015-10-19:
Hi,

unfortunately there is no workaround for now :(

SamuelBrucksch, 2015-10-20:
Hm too bad... So that chip is actually not a DS2433 but a similiar eeprom with SHA1 encoding. So the only way to write to it would be to find the key for encrypting the data. Would it help to monitor the onewire communication of a uprint to find that key?

bvanheu, 2015-10-21:
I don't think so, the data will be encrypted on the wire and decrypted by the chip. using the 64 bit key.

anubisis, 2015-11-15:
I have the same problem on the uprintse. If I understand correct, the chip is password protected only for write operations, not for read? Because when I read it, there is readable "STRATASYS" in ASCII, see images below. Also I noticed the chip family code is 0xB3 (no results on google).
[image]
[image]

My idea was to solder a DS2433 eeprom instead this password protected memory, but I guess that will not help.
So what about looking inside the printer and search for the encrypting algorithm?

HaveBlueXST, 2015-11-16:
The encryption is not handled by the printer - the encryption is handled by the reader in the material bay.  In theory, you could replace the material bay readers with ones from a Dimension (would need different contacts), and then you could use standard DS2433 chips.

makerswamp, 2015-11-27:
Hey Guys, I'm determined to get a solution for the uPrint SE+. I used the debug port on the back of the machine to grab the cartridge info.

Support Cartridge

    000000: b3 48 79 02 d0 14 10 da                           .Hy.....

    Support carrier in bay 0: 128 bytes at address 0 (128 8-bit values)
    000000: ae 5f c7 e6 3d c1 cb 67 be 08 10 4a 8a df c7 ed   ._..=..g...J....
    000016: 4e 76 5a 17 c6 2c 59 60 ce 7b e7 0a 20 b9 1f c5   NvZ..,Y`.{.. ...
    000032: a1 cc 9c 9f 21 df 19 b0 02 49 33 09 3f 6b 72 52   ....!....I3.?krR
    000048: 02 49 33 09 3f 6b 72 52 8b b5 92 42 7b ef 66 c9   .I3.?krR...B{.f.
    000064: 5f 7d 00 00 00 00 0d e5 55 aa 55 47 14 ed 64 be   _}......U.UG..d.
    000080: 48 76 00 00 00 00 00 00 83 7b 40 ed e1 92 1b e2   Hv.......{@.....
    000096: 3c cd b8 2e fc a9 f1 d2 53 54 52 41 54 41 53 59   <.......STRATASY
    000112: 53 92 fb 87 73 c0 33 37 45 5e c8 c5 cb d0 52 84   S...s.37E^....R.

Model Cartridge

    000000: b3 5c ef 03 d0 14 10 f9                           .\......

    Model carrier in bay 0: 128 bytes at address 0 (128 8-bit values)
    000000: f9 c8 d1 5f ad 37 08 d6 fe 88 b5 d3 2b 41 12 d6   ..._.7......+A..
    000016: b8 54 96 e0 da 12 df 6e f8 8b 38 9f b2 9d e0 e4   .T.....n..8.....
    000032: dc d1 c4 ea f2 95 1a 0f c3 3a d8 36 37 76 92 b9   .........:.67v..
    000048: c3 3a d8 36 37 76 92 b9 b5 ad 5c 76 01 60 37 91   .:.67v....\v.`7.
    000064: 87 c0 00 00 00 00 5b f9 55 aa 55 e3 21 83 5d be   ......[.U.U.!.].
    000080: 45 ee 00 00 00 00 00 00 db 34 af dc a9 d9 1f 18   E........4......
    000096: 08 fd d2 15 fc a9 f1 d2 53 54 52 41 54 41 53 59   ........STRATASY
    000112: 53 41 3e 6f 83 71 63 e6 bb cb b0 e1 82 38 9f 9c   SA>o.qc......8..

When I get home, I'm going to read the same data using the code from this repo (just checking everything works as expected).

My question is.... What could we do to create a 'smart' cartridge carrier that spoofs the machine? Could we remove the need for the one-wire chips all together and replace them with a microcontroller? or maybe a pizero? :-)

Would appreciate thoughts.

milesguidon, 2015-11-27:
Someone does have a solution for the uPrint SE+. Argyle Materials 
(Bolson Materials) have a spoof chip they make that goes in front of the 
contacts on the material/support hoppers.

makerswamp, 2015-11-27:
Here are the commands. Haven't tried the "Write" command yet....

[Material_Commands.pdf](https://github.com/bvanheu/stratasys/files/45963/Material_Commands.pdf)

makerswamp, 2015-11-27:
Hey Milesguidon,

Thanks for the info. I actually already have one of these. Do you know if this spoof-er works indefinitely? Argyle Materials still has a huge markup on their materials and it'd be nice not to swap one acumen approach for another.

milesguidon, 2015-11-27:
the point is that _they_ figured out how to spoof it!

makerswamp, 2015-11-27:
Agreed! I'll do some probing of the Argyle Materials DRD this afternoon.

bvanheu, 2015-11-27:
If anyone wants me to have a look on their device, feel free to reach me via my email address bvanheu@gmail.com.

makerswamp, 2015-11-29:
So, I've been doing some digging... I actually have a HP branded uprint se+. As such, the current code base cannot correctly decode my cartridges.

Seems the unique code in machine.py needed for my machine is different. 

    #
    # A machine is a printer from stratasys
    #
    type_to_number = {
        "fox": "2C30478BB7DE81E8",
        "fox2": "2C30479BB7DE81E8",
        "ktype": "6B2A268B5ED3374A",
        "prodigy": "5394D7657CED641D",
        "quantum": "76C454D532E610F7",
        "uprint": "F3A91DBE6B0B2255",
        "uprintse": "6B2A268B5ED3374A",
    }

Any ideas on how to find the code required for my machine?

256FF, 2016-08-05:
I'm playing around with and Arduino and the HP Designjet Chip. Reading is working well. What do I have to do to try to decrypt the HEX-File with the uprintSE machine number? Also I have tried to read the Bolson Materials DRD Chip but have had no luck. Maybe it is not a plain 1Wire chip. How can I figure out the machine number of the HP printer? Where can i "find" it?
I bought the DRD to try to use it with the HP Printer but the printer don't accepts the DRD -> Error. So I think there are differnces between the uprint SE Chips and the HP Chips. But in the DUMP I can see _I=ü©ñÒ**STRATASYS**.Å¡^©HT_ 

HEX-Dump of the HP Designjet 3D.
> device address = B3 33 6A 8 D0 14 10 93
> 
> EEPROM DATA (512byte) =(
> ED 25 66 25 3D 72 9D 2A 44 C7 A6 B8 35 03 E6 6D 
> 07 09 6C 52 5A FF DF 26 BE 2F 58 AF 17 3F 31 C9 
> 96 CF 84 D9 CD B1 BF CC 86 1B 56 24 D9 D4 A3 1B 
> 86 1B 56 24 D9 D4 A3 1B C4 9F 0A E4 51 81 83 AC 
> 1F AB 00 00 00 00 30 4A 55 AA 55 E7 FB 5E 51 BE 
> 1A 0C 00 00 00 00 00 00 AF 79 CE 87 30 3C 33 69 
> 1D 1D 49 3D FC A9 F1 D2 53 54 52 41 54 41 53 59 
> 53 18 C5 A1 5E A9 48 54 18 2F 5B E9 29 5B EC B3 
> FF FF FF FF FF FF FF FF 00 55 00 55 00 00 00 00 
> B3 33 6A 08 D0 14 10 93 FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
> )HEX

HaveBlueXST, 2016-08-06:
256FF, it looks like your EEPROM dump may be treated as a uPrint rather than a uPrintSE.  Here's what I got from it:

pi@raspberrypi:~/stratasys-master $ ./stratasys-cli.py eeprom -t uprint -e 931014d0086a33b3 -i odinchip.bin
Cartridge - 'odinchip.bin'
-------------------------------------------------------------------------------
Serial number           256934412.0
Material                unknown (511 - 0x1ff)
Manufacturing lot       6659
Manufacturing date      2014-12-16 15:14:50
Last use date           2014-12-16 15:14:50
Initial quantity        40.0
Current quantity        -1.16320953585
Key fragment            55aa55e7fb5e51be
Version                 1
Signature               STRATASYS

Machine type:           uprint F3A91DBE6B0B2255
EEPROM uid:             931014d0086a33b3
pi@raspberrypi:~/stratasys-master $

256FF, 2016-08-06:
Oh, I was thinking it is a SE Clone.
Thanks for your Analysis. This means, there are no differences in the cartridge. So normaly the uPrint cartridge can be used in the HP too. Right?

HaveBlueXST, 2016-08-06:
I'm not sure if a uPrint catridge can be used in the HP - you'd want to do a dump and analysis of a regular uPrint EEPROM and compare it.

Note that your dump showed an unknown material value of 0x1ff as well, which I've not seen before - perhaps that's how the cartridges are differentiated.

256FF, 2016-08-06:
A, good idea with the unknown Material. I will try to get some Dumps of an uPrint Chip. 
Once more .... Thanks to you.

256FF, 2016-10-31:
Here I'm back again. I recieved the DRD-Chip from Bolson and managed to read it out. Initial read worked well. Also the printer recognized the Chip als 922ccm. But after printing a few layers it stopped and it was no longer readable. I dumped again and tried to decrypt it but it shows the following error

> Traceback (most recent call last):
  File "stratasys-cli.py", line 252, in <module>
    app.run()
  File "stratasys-cli.py", line 27, in run
    args.func(args)
  File "stratasys-cli.py", line 101, in command_eeprom
    self._eeprom_info(args)
  File "stratasys-cli.py", line 211, in _eeprom_info
    cartridge = m.decode(machine_number, args.eeprom_uid, cartridge_crypted)
  File "C:\Temp\stratasys-master\stratasys\manager.py", line 52, in decode
    cartridge_packed = self.decrypt(machine_number, eeprom_uid, cartridge_crypted)
  File "C:\Temp\stratasys-master\stratasys\manager.py", line 202, in decrypt
    raise Exception("invalid current material quantity checksum")
Exception: invalid current material quantity checksum 

Here ist the dump, does anyone have an idea how to decrypt?
[1477898547.84.bin.txt](https://github.com/bvanheu/stratasys/files/561266/1477898547.84.bin.txt)

HaveBlueXST, 2016-11-03:
What is the EEPROM's UID?

256FF, 2016-11-03:
Sorry, forgot to add the UID

b30948e1d0141010

256FF, 2016-11-03:
Some more dumps from HP DesignJet Chips; Material is unknow 
Filename = UID (reverse)
[1e1014d006dcbbb3.bin.txt](https://github.com/bvanheu/stratasys/files/568541/1e1014d006dcbbb3.bin.txt)
[d31014d00b1c4bb3.bin.txt](https://github.com/bvanheu/stratasys/files/568542/d31014d00b1c4bb3.bin.txt)
[621014d006fc39b3.bin.txt](https://github.com/bvanheu/stratasys/files/568543/621014d006fc39b3.bin.txt)
[a01014d006b033b6.bin.txt](https://github.com/bvanheu/stratasys/files/568544/a01014d006b033b6.bin.txt)
[931014d0086a33b3.bin.txt](https://github.com/bvanheu/stratasys/files/568545/931014d0086a33b3.bin.txt)

and finally one dump of a support
[d71014d0086c11b3.bin.txt](https://github.com/bvanheu/stratasys/files/568548/d71014d0086c11b3.bin.txt)

256FF, 2017-02-07:
I finally managed to re-write the 100% dump back to the EEPROM of the HP. But this means that it is neccessary to delete the file with the stored serial numbers.
A detailed instruction can be found here: [https://www.3dprintforums.com/showthread.php/3153-How2-Refill-the-EEPROM-of-the-HP-DesignJet-3D-aka-uPrint](url)

thekakester, 2017-10-05:
Do you have a mirror to that link for the instructions?  That page doesn't exist anymore.
I used your arduino library and dumped my EEPROM easily (thanks for that!), but I'm having problems doing it from my pi 3 and I figured that article might shed some light on it.
I'm assuming you re-wrote using a pi, so I figured I'd go that route too.

modprobe w1_ds2433 creates a directory /sys/devices/w1_bus_master/b3-XXXXXXXXXXXX, and I can read the ID, but there's no file/directory labled "eeprom" like the readme mentions.

These are what I have to work with in that directory:  (d)driver, (f)id, (f)name, (d)power, (f)rw, (d)subsystem, (f)uevent

(where "d" stands for directory and "f" for file)
Is "rw" the new name for the "eeprom" file mentioned in the readme?

bvanheu, 2017-10-05:
@thekakester : the reason why you don't see the "eeprom" file is because there is no driver that supports the B3 class in Linux. I'm currently working on such driver to support the MAC thing for the B3 family chips. The B3 family supports the same command to read the memory data as the 23 family, so you can use an arduino or bus pirate to dump the eeprom.

thekakester, 2017-10-05:
@256FF Were you able to write back to the EEPROM using the arduino for your b3 chip?

256FF, 2017-10-06:
Hi, 
i've found not solution for rewriting the eeprom with anything other than via DIAG port. I heard that maybe there will be an opportunitiy in the next weeks.

thekakester, 2017-10-06:
@256FF You are truly amazing and I have no idea how you found out that you could interface with the printer itself like you did.  I just saved the 100% data from some of my cartridges, and now I'm doing a print job to run the spool empty and I'll try to restore the 100% data back to it.  Thanks again for your time!

256FF, 2017-10-06:
Thanks for the flowers, but nealy nothing of this was found out by me. :-)
Take care, that the eeproms will have at least 1% left. If it is empty the eeproms will be write protected and you can't flash your dump.

ericalbers, 2017-10-27:
I have a uprint SE, and I have a image of its hard disk, I see a sha1 file on one of the volumes, could this be the encryption key used?

ericalbers, 2017-10-27:
Heres the file, its exactly as on the partition just in case its useful

[sha1.zip](https://github.com/bvanheu/stratasys/files/1423078/sha1.zip)

ericalbers, 2017-10-27:
Given the B3 is not in the list of onewire family chips, its probably a DS28E01, which doesn't have a family designation, uses the same write/read commands
https://www.maximintegrated.com/en/products/digital/memory-products/DS28E01-100.html

ericalbers, 2017-10-27:
I've found a file 'masterd' which has strings of system.dat in it, and strings like
BAD_CASS_REUSED and BAD_CASS_TARGET, BAD_MX_CONDITION
BAD_OUTSIDE_CRC_MATERIAL , BAD_INSIDE_CRC_MATERIAL  BAD_OUTSIDE_CRC_KEYFRAG

ericalbers, 2017-10-27:
three chip downloads for Material from a uprint se printer carts

[uprintsechips.zip](https://github.com/bvanheu/stratasys/files/1423367/uprintsechips.zip)

ericalbers, 2017-10-27:
Link to 'Side sha1 attacks on the DS28E01 eprom series'
https://www.cs.bham.ac.uk/~oswalddf/publications/cardis15_slides.pdf

tjschweizer, 2018-06-11:
@ericalbers 
How did you did an image of the uPrint hard drive?

Can you share that masterd file? And possibly a file list?

The Fortus 450 has the same B3 family code.

bigrom26, 2021-06-23:
Hello 
Is the topic topical.
How can you put the eeprom contents new and after use.
Take a picture of the head

trickd122, 2025-01-07:
> Hey Guys, I'm determined to get a solution for the uPrint SE+. I used the debug port on the back of the machine to grab the cartridge info.
>
> Support Cartridge
>
> ```
> 000000: b3 48 79 02 d0 14 10 [...]

Not sure if youre still on here however how did you connect to the debug port? What OS and whats the hardware setup? Im having difficulty connecting.