bvanheu/stratatools #22: Bus Pirate to read/write Uprint

jthand, 2015-05-27:
Hello,

I have made an attempt to read the chip using a bus pirate. I think I have it connected correctly. I have downloaded python and pycrypto. I am using PuTTy to connect to the Bus Pirate, when I enter  "$ ./bp_read.py /dev/ttyUSB0 eeprom.bin" to read the chip, all that comes up is BOOTLOADER. Would someone please school me a bit to tell me what I am doing wrong?

Thanks,
John

rowlandski, 2015-05-28:
Can you give a bit more detail?
Is the eeprom.bin file generated? If so, what's its contents?

R

jthand, 2015-05-28:
Sorry,
When I use the Putty I get the following:

1-WIRE>$  ./bp_read.py
Are you sure? y
BOOTLOADER
It does nothing after the BOOTLOADER and I have to reset the Bus Pirate. I have followed the directions from other examples and tried to do it manually and get the below:

SEARCH (0xF0)
Macro     1WIRE address
 1.0xB3 0xEF 0xEF 0x00 0xD0 0x14 0x10 0x2F
   *Unknown deviceDevice IDs are available by MACRO, see (0).
the r:512 results in a repeated 0xFF

RobListon, 2015-05-28:
It looks like you are using putty to connect directly to the bus pirate and then entering the python command. You should just be running the python command which will open the connection to the bus pirate and run the commands for you.
The "1-WIRE>" prompt means that you are at the bus pirate's 1-wire mode command interface.

bvanheu, 2015-05-28:
Exactly as Rob put it, you just have to connect the bus pirate to your device, then run the Python script. The first parameter is the path to the serial device, the second is the path where the EEPROM should be dump.

It might be easier if you're booting into Linux.

jthand, 2015-05-28:
Thanks,

I am not at all familiar with connecting to a com port with python. Is there any tricks? When I copy and paste the "$ ./bp_read.py /dev/ttyUSB0 eeprom.bin" I get a syntax error. Would you please give me some basic how to's

RobListon, 2015-05-28:
1) Go to your "/stratasys/helper" directory (For me it's "~/stratasys/helper").
2) Type "./bp_read.py /dev/ttyUSB0 eeprom.bin"
3) Profit! No, just kidding. It should dump the eeprom to a file for you in the helper directory. If it doesn't it should give you an error with traceback that can help track down the problem.

jthand, 2015-05-28:
Rob, Thanks for the help. I am doing the following, but I am getting the following.

>>> os.chdir('C:\Users\jthan_000\Desktop\stratasys-master\helper')
>>> ./bp_read.py /dev/ttyUSB0 eeprom.bin
  File "<stdin>", line 1
    ./bp_read.py /dev/ttyUSB0 eeprom.bin
    ^
SyntaxError: invalid syntax

RobListon, 2015-05-28:
Oh, you are running this from a windows machine.
Then you would type "bp_read.py COM1 eeprom.bin" with COM1 being replaced with whatever com port the bus pirate shows up as.

jthand, 2015-05-28:
That helps but I am now getting the following:
>>> bp_read.py COM4 eeprom.bin
  File "<stdin>", line 1
    bp_read.py COM4 eeprom.bin
                               ^
SyntaxError: invalid syntax

RobListon, 2015-05-28:
I just installed my bus pirate on my windows machine to try and see if I could get the same error but can't replicate it yet. I think you will need to post the entire command you typed and it's output.
Also, can you try "python bp_read.py COM4 eeprom.bin" and see if you get a different result?

jthand, 2015-05-28:
I have switched to fedora at the moment. I am in the file directory but I get:

>>> os.chdir('/home/johnhand/stratasys/helper')
>>> ./bp_read.py /dev/ttyUSB0 eeprom.bin
  File "<stdin>", line 1
    ./bp_read.py /dev/ttyUSB0 eeprom.bin
    ^
SyntaxError: invalid syntax
>>> bp_read.py /dev/ttyUSB0 eeprom.bin
  File "<stdin>", line 1
    bp_read.py /dev/ttyUSB0 eeprom.bin
                                                        ^
SyntaxError: invalid syntax

RobListon, 2015-05-28:
What version of python are you running? I think Fedora uses python3 as the default now and this program doesn't run in python3 yet.

jthand, 2015-05-28:
It is 2.7.8

RobListon, 2015-05-28:
Hmm...how are you getting the exact same syntax error on both windows and linux? Are you running these commands directly from a terminal window on either of these operating systems?

jthand, 2015-05-28:
I am running out of the terminal window on fedora and the cmd window in Windows. I know I have to be missing something small, but I can't pinpoint it. Below is the terminal code and error I am repeadly getting:

[johnhand@localhost ~]$ python
Python 2.7.8 (default, Nov 10 2014, 08:19:18) 
[GCC 4.9.2 20141101 (Red Hat 4.9.2-1)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.chdir('//home/johnhand/stratasys/helper')
>>> bp_read.py /dev/ttyUSB0 eeprom.bin
  File "<stdin>", line 1
    bp_read.py /dev/ttyUSB0 eeprom.bin
                                                        ^
SyntaxError: invalid syntax

bvanheu, 2015-05-29:
It seems like `bp.onewire_read(512)` does not return the EEPROM data correctly.

Please see the issue #8 . You can always dump the EEPROM manually, but if you are able to debug & fix the problem on Windows, that would be great ;)

jthand, 2015-06-11:
I tried issue #8 already but all I get is a repeated 0xFF. I have gotten the bp_read.py file to run with no issues (I think). It creates a bin file that has a size of 1kb. Could you give a newbie some direction on how to read the file?

The link to the file is below:
 https://drive.google.com/open?id=0B_hWDzwnVoPmYzNHQW1mRFhKSUE&authuser=0

bvanheu, 2019-06-30:
Wont fix as the rapsberry pi is the preferred method to read/write on the eeprom.

closed 2019-06-30 by bvanheu