bvanheu/stratatools #23: Can Dimention or Fortus printers be used?

strozster, 2015-07-30:
Can Dimention or Fortus printers be used? I have the .bin file that I copied from the device. Is it possible to view the contents of it like other printers? Right now I am getting errors running the stratasys-cli.py eeprom -t command.

bvanheu, 2015-07-31:
Yes you should be able to use this software to parse Dimension & Fortus printer cartridge.

Please tell me the exact line you're using to parse the `.bin` file that gives you an error, or I cannot help you.

You should use something similar to:
```
$ ./stratasys-cli.py eeprom -t fox -e 4343434343434343 -i random_file.bin
````

Make sure to replace `fox` by the correct machine type, `4343434343434343` by the EEPROM id and `random_file.bin` by your filename.

strozster, 2015-07-31:
Attached is a screen with 3 commands I ran on a eeprom.bin file that I retrieved. It seems I get an error with "dimension" as there is no option for it. When I tried the same thing replacing the machine with fox it gave several errors. 

[image]

bvanheu, 2015-07-31:
Try with each of them until you have a correct decryption.

Also, make sure the EEPROM UID ends with `23`. Yours seem to start with 23. You should reverse it.

strozster, 2015-07-31:
for example if my eeprom uid reads 23a29e1g10222272 what should I be typing in? a29e1g1022227223? or 27222201g1e92a32?

bvanheu, 2015-07-31:
`23 a2 9e 1g 10 22 22 72`
should be:
`72 22 22 10 1g 9e a2 23`

(without any spaces of course).

there should not be any `g` in there though, make sure you're using a valid EEPROM id.

strozster, 2015-07-31:
Below are screens of the output for a script I wrote to go through all of the options. Errors were present for all. There is no g I flipped a few characters. If you need anything else let me know.

Thanks for all the help. 
[image]
[image]

bvanheu, 2015-07-31:
The wrong checksum means the decryption was not successful, probably because the wrong key is used.

People in the past have successfully decrypted EEPROM for Dimension printers, so make sure the EEPROM dump is correct and you're using the right EEPROM uid.

If you're *pretty confident* you did everything right, you can send me your EEPROM + the EEPROM uid by email (see in the README) and i'll see what I can do. You might have an exotic printer model ;)

toma678, 2015-08-08:
Any update on this? The EEPROM dump I have is throwing checksum errors on every machine type listed. I can send you the dump and ID if possible. I'm running a Stratasys Dimension SST 768.

RobListon, 2015-10-12:
I am currently using this project with both my Dimension and Fortus machines. It works great.

573F4N077, 2016-03-30:
@RobListon 
what is the model name you are using? (prodigy or fox or...) thx

RobListon, 2016-04-05:
I am using the prodigy type for both the Dimension and Fortus.

strozster, 2016-09-20:
@bvanheu I know your comment is over a year old, am I still able to send you an email with the eeprom dump for the printer I am trying to use? I have it working for the Dimension but not the Fortus (I'll specify the model in the e-mail), I continue to get the checksum error. Thanks.

bvanheu, 2016-09-20:
sure.

strozster, 2016-09-28:
Sent you an email, let me know if you get it! Thanks

LUKIBOB, 2016-10-13:
I have the eeprom from a ABS-M30 canister that goes in Fortus360, 400,... Will this code work for it?
[image]

HaveBlueXST, 2016-10-13:
LUKIBOB, yes, the code will work with it (there's no MAC authentication to deal with, unlike later chips).

LUKIBOB, 2016-10-16:
Thanks HaveBlueXST, I just got a Titan TI and have near limitless supplies of canisters. Need to refill them and hopefully find a material setting with the correct temperatures. Wish I could set the temp and print speed manually like Repetier Host. I'll be working on that next

closed 2016-12-02 by strozster