bvanheu/stratatools #24: Illegal Cartridge Error...

rowlandski, 2015-08-13:
Morning,

I've read, edited and re-programmed an EEPROM as below. But when trying the cartridge with the re-programmed EEPROM I got an illegal cartridge error.

Read EEPROM (empty):
************************************************************************************************************************
Serial number		220078410.0
Material			SR30 (171 - 0xab)
Manufacturing lot	5817
Manufacturing date	2014-04-03 12:15:06
Last use date		2014-04-03 12:15:06
Initial quantity		56.3
Current quantity	0.0626293700581
Key fragment		55aa556bbdc355be
Version			1
Signature			STRATASYS

Machine type:		prodigy 5394D7657CED641D
EEPROM uid:		290000016c560823
************************************************************************************************************************

Re-programmed EEPROM to:
************************************************************************************************************************
Serial number		220078415.0
Material			SR30 (171 - 0xab)
Manufacturing lot	5817
Manufacturing date	2014-04-03 12:15:06
Last use date		2014-04-03 12:15:06
Initial quantity		56.3
Current quantity	22.2
Key fragment		55aa556bbdc355be
Version			1
Signature			STRATASYS

Machine type:		prodigy 5394D7657CED641D
EEPROM uid:		290000016c560823
************************************************************************************************************************

Note:

I changed the following:

Serial number from:       220078410.0            to   220078415.0        (increased by 5)
Current quantity form:   0.0626293700581    to    22.2                      ( new number)  

Any ideas why I get an error?

Should the, current quantity 22.0626293700581 to keep it in the same format?

Any help would be appreciated.

Regards,

R.

bvanheu, 2015-08-13:
I've never seen an "illegal cartridge error" before, but you can try keeping the spec of the cartridge as close to the original as possible.

What is your printer? Does it used to work before?

I don't own a printer myself, but maybe others might be in a better position to help you.

slaytonrnd, 2015-08-14:
I think we always got that error when trying to reprogram a support material cartridge and we never got around it.

Are you trying to refill support material?  Last time I checked (at least a year ago) we couldn't find refill filament for the support material we were using, but maybe SR30 is available.

Are you wanting to use the support material cartridge for model material?  It is probably possible to change the material type for the cartridge from support material to model material and use it, but we've never tried it, so don't know for sure that it would work.

rowlandski, 2015-08-18:
Thanks for both of your replies.

I don't have a printer, but i'll be gaining access to one in the next few days.

Regarding support material, I initially only have one EEPROM PCB, so just started with what I had. It happened to be support material (didn't know the difference at the time), but I just wanted to see if the "update" EEPROM would be accepted. And from that test, it seems the format of Current Quantity has to be correct.  

I have re-programmed four EEPROMs, which I'll be testing soon. I will publish the results in due course as it may help others.

One question I do have, not actually seen a printer cartridge, but the Initial Quantity is set to 56.3. Any idea what the units are?

Regards,

Rowland

HaveBlueXST, 2015-08-18:
Rowland, the units are in cubic inches.

rowlandski, 2015-08-28:
Managed to actually test various re-programmed EEPROM and had access to a printer (Fortus).
It seems that re-programming Current Quantity with say 23.6, will make the system fail.
The value either has to be set to the max, in this case, 56.3. Or if a non-full value is required then all the decimal fractions of an integer need to be added (not sure of the correct word, so here’s an example:
23.9		                Wrong
23.9256486723	Correct
When reading a cartridge all the digits are shown if the cartridge is not full, so when re-programming with a non-maximum value it requires all these digits.
So, to sum up it worked. Just doing so further tests.
Regards,
.
Note: The serial number of the cartridge always has to be increased indicate fool the printer in thinking it has a new cartridge.

Moffman76, 2016-02-03:
By changing the serial number, does this mean that you don't need to remove system.dat on boot?

Moffman76, 2016-02-03:
Update: We fixed a cartridge we thought we bricked! Roll serial number forward, and change material quantity back to 56.3. Thanks for the info guys!! Cartridge Writer is awesome.

closed 2016-02-23 by bvanheu

trickd122, 2025-01-07:
> Morning,
>
> I've read, edited and re-programmed an EEPROM as below. But when trying the cartridge with the re-programmed EEPROM I got an illegal cartridge error.
>
> Read EEPROM (empty):
>
> Serial number [...]

Hello. Did you ever solve this? Im having difficulty connecting to the chip. Arduino uno windows 10. Cartridgewriter wont read and just closes after hitting run