bvanheu/stratatools #26: Question: Key Fragment

hahathejellyfox, 2015-12-04:
Hello, first of all I would like to say thank you for your wonderful software. I am planning to use it soon and was wondering what I should enter for the key fragment argument when creating a new EEPROM file.

Cheers,
Danny

bvanheu, 2015-12-05:
Hi,

You can put whatever you want for the key fragment. The format is 8 bytes,
hex encoded.

e.g.:
0123456789abcdef
4242434445464748

The key fragment is used to build the key that will be used to both encrypt
and decrypt the EEPROM.

Benjamin Vanheuverzwijn

hahathejellyfox, 2015-12-08:
Thanks :) I was also wondering what the initial material and current material fields are for

bvanheu, 2015-12-08:
`initial material quantity`: the quantity of material in cubic inches initially in the cartridge
`current material quantity`: the remaining quantity of material in cubic inches.

When `initial material quantity` == `current material quantity`, it means that the cartridge is full (100%).

hahathejellyfox, 2015-12-09:
Excellent thanks for all the help

closed 2015-12-09 by hahathejellyfox