bvanheu/stratatools #27: why the cartridgewriter can't work?(uprint)

Legendrechuang, 2016-01-05:
I follow the step in README.md. Install "teensyduino" and add "onewireproxy. ino" in Arduino. and Wire Arduino to DS2433 chip. but after Press the Read button, "CartridgeWriter. exe" just crash, why? My way to differentiate ground pin and data pin of Uprint's chip is form this issues: bvanheu/stratasys#21
And the photo is that how I connected.
[image]

slaytonrnd, 2016-02-03:
Short answer: I (slaytonrnd) have not had time to update CartridgeWriter with the latest code from bvanheu/stratasys, so I'm not sure it will work with the uprint.  I'm not sure at this point when I will get a chance to work on CartridgeWriter so I don't want to make any promises on when it will be functional for uprint.

bvanheu, 2016-02-23:
thanks @slaytonrnd  for the follow-up.

@Legendrechuang The UPrint chips are unfortunately not supported yet.

closed 2016-02-23 by bvanheu