bvanheu/stratatools #29: Exception: cannot copy scratchpad

didonatoandres, 2016-02-23:
Hi.

By using BusPirate, I've managed to read the cartridge and decode the information.
Then I tried to generate a new binary for my cartridge.

But when I try to write it, I get the following exception:
Exception: cannot copy scratchpad, scratchpad is (binary bytes, ending in many 0xFF's)  but i sent (binary bytes, none of them is 0xFF's)

Have you seen this happening before? Thanks

didonatoandres, 2016-02-23:
I add some new information:
By using the argument -p, I tried to replicate the original cartridge. Then, I compared the original hexdump with the new hexdump. Surprise: different. Suprise #2: trying to write the original or the replicated binary file, they both lead to the same exception cited above: cannot copy scratchpad, scratchpad is.............

bvanheu, 2016-02-23:
hmm I never heard of someone having this problem before.

What is the family code of the chip you are using?

didonatoandres, 2016-02-23:
It doesn't seem to be a commercial IC, yet it responds perfectly to 1W.
Code is made up of three fields, set up in rows:

331101
0905B1
020AC

It has two wires, just as Dallas/Maxim commercial DS2431

It is mounted on a plastic case as shown in the image (Model is uprint).
[image]

New things I've been doing:

a) I've tried another cartridge, and result is the same, so it is not a particular cartridge.

b) I could write the cartridge using the serial diagnostic terminal on the printer, but there's a limit: apparently after six write cycles, writing is not allowed and the error message is: "
"PClassMaterial.cpp:163:INFO:Cartridge communication error:material write fail limit of 6 exceeded on Model cartridge"
I can't tell if this "count" is being made by the cartridge itself or by the PC. I'd bet it's the former.

c) Further testing: insisting on writing with 1W through Bus Pirate, I decided to ignore the exception editing the Python script, and tried to write the whole memory. It can be seen that after writing, the memory content is the same that it had previous to be written. So actually no bytes are being written at all.

All these make me think that this IC is a little more complex than the plain old 2431.

fsabbatini89, 2018-03-01:
Hi didonatoandres,
did you manage to write to this memory?

didonatoandres, 2018-03-01:
Hi.
Not really, but I could find out a little more.
Just us I thought, uprintSE cartridges are not exactly a DS2431, they're something more similar to this:
https://www.maximintegrated.com/en/products/digital/memory-products/DS28E01-100.html

While writing through the "DIAG" serial terminal, writing was limited to 6 times.

Now, writing directly to the IC was not possible to me.
This guy uses a SHA engine, so even though reading works just as the usual 1Wire protocol, writing is a bit more complicated. I guess that everything you need to solve this thing is inside the hard drive of the printer, but the thing is our printer just died for other reasons, so I've lost the interest in working on it.

Good luck.