bvanheu/stratatools #31: Copying eeprom data using the buspirate

strozster, 2016-06-22:
Using buspirate 3.6 I was getting errors that there was no one wire device connected when I ran the bp_read.py file (error was from the bus pirate). As per issue #8 I tried doing it manually using the buspirate com port commands, all I recived back was 0xFF, this was verified by scoping the line. I am using python 2.7.12 pn windows and python 2.7.9 on linux, neither work, and I don't expect them too since I can't even get it working manually. 

The pullup is enabled to 5v, when I connect the eeprom it drops to 1.72V (This is under the voltage as per the ds2433 datasheet) which I was concerned about so I tried using an external 4.7k which didn't work and a 2.2k, which didn't work (line voltage was above threshold at 3.3V and 2.8V respectively) all I was getting on the scope was the line was high (no drop in voltage and read starting to happen).

HaveBlueXST, 2016-06-22:
The only way that I've had any luck with using a buspirate to read/write an EEPROM is by cutting the trace to the pulldown resistor on the EEPROM PCB.  I've had excellent results with using a [RasPi and a 2.2k resistor](http://haveblue.org/?p=1988), so that's my current preferred method.

strozster, 2016-09-02:
Got it working using pi.

closed 2016-09-02 by strozster