bvanheu/stratatools #35: Windows 7 ImportError: No module named cipher

LUKIBOB, 2016-10-19:
How do I get rid of this error? I have tried many things, install, uninstall from python pycrypto change cipher to Cipher, what is the correct way to do it? I am running raspian on a pi3 and windows 7 on the pc.  I cannot get the command
"python stratasys-cli.py eeprom -t fox -e 8e0000020fe48423 -i cartridge_dump.bin -r"
to work.
It says     from crypto.cipher import DES, XOR
ImportError: No module named cipher

LUKIBOB, 2016-10-19:
8e0000020fe48423

bvanheu, 2016-10-19:
You need to install pycrypto:
http://www.voidspace.org.uk/python/modules.shtml#pycrypto

LUKIBOB, 2016-10-20:
I have installed pycrypto 50 times, on win7 and raspberrypi. I cannot get it to stop saying "no module named cipher"  
On my pi the latest attempt it says "Successfully installed pycrypto"
Is my path bad for the stratasys-cli file?
/usr/lib/python2.7/stratasys-master/   Do I need to move that folder or stratasys-cli.py file somewhere else?
[nopycryptolog.txt](https://github.com/bvanheu/stratasys/files/540839/nopycryptolog.txt)

LUKIBOB, 2016-10-20:
I have another computer I got pycrypto to work but am getting 

raise Exception("invalid content checksum: should have " + hex(struct.unpack
("<H", str(cartridge_packed[0x40:0x42]))[0]) + " but have " + hex(self.checksum.
checksum(cartridge_packed[0x00:0x40])))
Exception: invalid content checksum: should have 0x68e7 but have 0x1af5

I have the eeprom.bin from the raspberrypi gpio dump, but cannot attach it here

[8e0000020fe48423.txt](https://github.com/bvanheu/stratasys/files/541899/8e0000020fe48423.txt)