bvanheu/stratatools #36: Machine numbers of upritnse and ktype are the same?

256FF, 2016-10-19:
Is is true, that the machine numbers the machine numbers are the same?
https://github.com/bvanheu/stratasys/blob/master/stratasys/machine.py#L11-L15

`"ktype": "6B2A268B5ED3374A"`
`"uprintse": "6B2A268B5ED3374A"`

bvanheu, 2016-10-20:
Both machine numbers come from different firmware under those names, so I left that as-is. I'm not sure what is a "k-type" machine.

Silvood, 2017-02-21:
If I have correctly understood the way Stratasys electronic hardware works, the soft on the board is the same for all FDM printers, and can be edited on the machine. 
I saw a gender: neutral in the code which reffers to a board not programmed, so maybe the k-type is related to this, a board/chip wich hasn't been yet programmed?

bvanheu, 2019-06-28:
UprintSE might use this id: `09FBD4B61FC0B327`

As soon as anyone confirm it works i'll close all related bugs to UprintSE :)

Belencsakl, 2020-04-06:
TO 256FF
Hi Odin, 
I found your project: DS2433 Uprint emulation.  This is work on the "laboratory", But if I put in to printer. Printer Not recognize the Eprom. I killed my original EProm, and I now search a solution. 
Work this project? Or this is only for test?
Thank you!

256FF, 2020-04-06:
Hi Belencsakl, 
unfortunately I did not succeed in finishing an emulator for the uprint. 
If your EEProm is wrote protected your only chance is to buy a new one. To refill it, it must have at least 2% left  on the chip.

Belencsakl, 2020-04-07:
Hi Odin,
thanks for your response. I'll work a little on project. I'll let you know, if I have any news. 
Yes, the "normal" refill work. But I changed HDD in the printer, and forget change the serial number on HDD and by boot killed the eprom in Printer:(

This is my another project:) Printing  to PEI without base material but with brim.

[image]

[image]