bvanheu/stratatools #37: Cannot read eeprom with raspberry pi

mayrthom, 2016-10-25:
Hi,

i tried to read out the eeprom of a stratasys uprint with a raspberry pi, but it didn't worked as i had expected.
I wrote "dtoverlay=w1_gpio,gpiopin=4" in the file "boot/config.txt" and i can also see  the device in the folder "sys/bus/w1/devices/w1_bus_master1"
But in the folder "sys/bus/w1/devices/w1_bus_master1/b3-xxxxxxxx"  there is no file called eeprom. Just driver , id, name, rw  subsystem, uevent.

Does anybody know this problem?

Regards 

Mayr

Silvood, 2017-01-16:
Hello,

I'm having the same problem here on a rasberry pi 1 model B
1Wire activited via raspi-config, everything works to connect to the chip and read the eeprom's ID (b3-... I have a HP DesignJet=>so basically a Uprint chip but with a different ID if I understand correctly).
But no file or directory "eeprom" in the devie directory, same as @mayrthom :(

As many people has been able to read the eeprom, I think there is something we both done wrong :/ Anyone to help us?

Thanks,

Silvood

256FF, 2017-01-21:
I own the HP too. There are some things to do. Eeprom size ja only 128bit. I managed to read ist with an Arduino an Windows but writing did not Work. 
Now I so everything with rs232 in the DIAG Port. I will write a how2 at 3dprintforum.com in the next week.

Silvood, 2017-01-31:
@256FF , I would be happy to have a how2 on how to use the diag port for this :)
I think you can easly read, but can you also write on chips with the diag port? I have actually 5 HP material chips with near to 0% for testing purpose :)

256FF, 2017-01-31:
writing is as easy as reading with diag port and hyperterminal. BUT: To let the printer accept the re-written 100% eeprom you need to delete the "Serial-number-memory-file". As e.g. show for the Dimension 768 with edited rc.local at every startup. I must confess that I do not have much time atm to write the how2 in the next few days but it will come soon at 3dprintforums.com

256FF, 2017-01-31:
1. d

dfdf

dff

1. ss

256FF, 2017-02-07:
@Silvood 
I finally managed to re-write the 100% dump back to the EEPROM of the HP. But this means that it is neccessary to delete the file with the stored serial numbers.
A detailed instruction can be found here: [https://www.3dprintforums.com/showthread.php/3153-How2-Refill-the-EEPROM-of-the-HP-DesignJet-3D-aka-uPrint](url)

HaveBlueXST, 2017-02-10:
@256FF , do you mind if I post a copy of your .pdf instructions on my [blog post](http://haveblue.org/?p=1988)?  Many people have asked me about how to hack the uPrint series.

256FF, 2017-02-10:
Of course not. I would be glad if you post my instruction in your Blog.

HaveBlueXST, 2017-02-10:
Thanks!  Could you also send me an email when you get a chance?  I had a few questions for you.  You can reach me at haveblue@haveblue.org.

Silvood, 2017-02-17:
@256FF Thanks a lot for the how2!
I'm about to unplug the harddrive today, apply the modifications this weekend (with a backup before :3) and test it tuesday when I'm back at work.

256FF, 2017-02-21:
@Silvood 
How are you doing? Worked the modifikation on your printer?

Silvood, 2017-02-21:
@256FF Yes, just finished and applied some more :)

As I didn't wanted to wait the weekend to bring home the hard drive, I didn't backup, and directly edited rc.local to delete system.dat on startup. (I was pretty confident that nothing could go wrong at this stage)
I will began material test as soon as I have new Stratasys cartridges (maybe at the end of the next week).

Friday wasn't over, and nothing else appeared at work, so I decided to try transform the HP to Stratasys.

I did that cause I didn't find any HP cartridge reseller (even HP itself...) and I don't have any support cartridge left... But I know a Stratasys reseller who can sold me some so I can bypass them in the future 

After 2 days of labor, I have successfully apply all the modifications. So my HP is now a full and real stratasys Uprint (I can also go back to a full HP if needed)
I am currently writing a how2 (in French first, English will come :) ) for this. A quick explanation could be: editing gender in /mariner/config/machine.cfg, then use the diag port and rg/sg command to edit gender, then edit the serial number in a specific eeprom (mr 0 1 0 123). Finally, update the firmware.

Now, HP cartridges are "material not recognize" (the same way Stratasys' are on a HP) but Stratasys cartridges will be (I am 95% sure ^^)

I hope it can help some for creating chip without dump ^^

256FF, 2017-02-21:
Wow, you switched from HP to uPrint. Perfect. I'd like to know all about this transformation.

Silvood, 2017-02-21:
I'll do the same thing as you did, posting a thread on 3dprintforums.com with the how2 :)

@mayrthom, to return to the original subject, the best way to read Uprint/HP eeprom is to use the diag port as described in the how2 of @256FF , cause the chip isn't the same as other stratasys' printers.
I tested it and it's way faster than connecting wire to the chip ^^

256FF, 2017-02-21:
Would be great. I'm looking forward for your next posts .... With success!

256FF, 2017-02-22:
I can't wait ;-D

Why do I have to edit a serial number of a specific eeprom? And what is the mr command? I have not found any information about this command in the terminal.

Below is the machine.cfg of my HP.
Changing the Gender form **LAHAINA** to **PAIA** will make my HP Printer to a **uPrint Plus**.
Do I need to change the **_SERIALNUMBER_** and **_UDN_** in this file too?

By the way, changing HANA to PAIA will transform the uPrint to a uPrint Plus, right?

- ZAXISHOMEOFFSET=-0.053005
- XYAXISID=UNKNOWN
- YLASH=0.000000,0.000000,0.000000,0.000000
- GENDER=LAHAINA
- XCORR=0.000000
- SERIALNUMBER=M00***
- PCAL=5.519500,5.518500,7.514000,7.512000,9.315000,9.327500
- UDN=is1M00***
- UPNPCONFIG=1
- LANGUAGE=de_DE
- OVERRIDE=0

Silvood, 2017-02-22:
Yes, you can make it a Uprint+, but be aware that differences between uprint and uprint+ are material AND build size, so you can’t upgrade a uprint to a uprint+, that’s useless (except for different material, but we can bypass it ^^)

 

The “specific eeprom” as I called it is the FrameIDMemory as called by stratasys. It stores many information of the printer.
To understand clearly what I’ve done: There is 2 boards which control the machine: the computer and the main board attach to it. At each startup, the boot process compare the gender and the serial ID of both the computer and the main board.
Changing the gender in machine.cfg automatically update the serial stored in the computer at the next startup (UDN is simply a concat of gender+serial)

So, to boot correctly and don’t have the 15.1 error, we need to update these informations in the main board. 

 

Rg/sg are made to read and change the gender in the main board.
mr and mw are made to read/write the FrameIDMemory (memory in the main board) => you can type “help” to list all commands, and “help mr” “help mw” to understand each commands :)

In the log, I saw that the error started when comparing a portion of the FrameIDMemory (0 1 0 123 If I remember), so I read it, saw the Serial, and edit it to the new.
You can try not to edit the FrameIDMemory but restore the original serial in the computer (keep the actual (mine was M00255 with HP, switched to P00255 with stratasys, so I should have write back M00255) and test. If it works, it’s cool, but I’m not sure.

I think I’ll be able to finish the whole how2 before the end of the next week (3rd march)

Be careful with what you’re doing, as changing the gender force a firmware update ^^ (I’ll posting Stratasys’ lastest firmware with the how2)

256FF, 2017-02-22:
Hmm, if there is a firmware update maybe it is not possible to revert back to a HP printer. I first thought I will have two harddrives, one for uPrint and one to restore HP easily.
But changing the harddrive will not be enough, right. I need to change sg and mr to let the printer accept the other harddrive - but it is still unclear if we can get back to HP after the firmware update was done.

Silvood, 2017-02-22:
I actually did not try when fully converted, but I had to revert back when I wasn’t understanding the FrameIDMemory modifications.

 

What I have done:

Edit machine.cfg in SBC with gender HANA (uprint) (SGC: Single Board Computer, as named in Service Manual),

Edit gender in CB with rg/sg (9:Uprint) (CB: Controller Board)

Reboot

Update with stratasys firmware

reboot

Seeing it doesn’t boot (error 15.1)

 

Edit gender back to LAHAINA both in SBC and CB (error 15.1 does not block accessing to diag port or SBC)

Reboot

Update with HP firmware

Reboot

Booting correctly.

 

I think that editing the FrameIDMemory won’t change at all the process, so that we can roll back when we want.

 

FIY: the last Stratasys update for Uprint is 9.1B3550, same as HP: 9.1B3550 (But I won’t try update a stratasys gender with a HP update nor HP with stratasys update.)

 

I’m starting a new thread in 3DPrintForums for this :) (in /stratasys)