bvanheu/stratatools #40: New Black Eeprom DS28 for Stratasys, any solution ?

ciboulete, 2017-01-04:
Hello Everyone, 

Well, I'm writing this post because we recently bought a Stratasys Fortus 450 and what a surprise, the cartridge have a new black eeprom with the slot on the side instead of on the middle with green ones....

We tried to read them, but impossible to communicate with, it seems to be a DS28E01 Eeprom. Did you already see this kind of eeprom and were you able to read it ?

If you have any information, please do not hesitate to contact me.

Thank you all.

Ciboulete

Mickele84, 2018-02-15:
News??

Mickele84, 2018-02-28:
Thank you so much

SKLM2771, 2018-03-12:
Hello, I also searching for solution of this problem. I think that it is possible to buy DS28E01-100, write on it and and place it on the chip, what do you think?

Mickele84, 2018-03-12:
Hi

Good idea!! 💡 

Saluti

Michele Rossetti

Inviato da iPhone

Mickele84, 2018-03-13:
Hello bvanheu,

I ask you for information for cartridgewriter for uprint, i can not read internal eprom the cartridge, what can it be?

Thanks.

Saluti

Michele Rossetti

Inviato da iPhone

glenoman, 2021-12-30:
hi, any updates on this? I cannot decode/write new black tiny eeproms for F900. I can see the crypted eeprom content but all machine types gives me "invalid crypted content checksum" error when I try to decode.