bvanheu/stratatools #41: Material not identifiable

256FF, 2017-02-21:
I dumped a lot of chips of my HP Designjet (aka uprint) and can read the values. But Material is shown "unknown (4095 - 0xfff)". Can anyone please help me to add my Material to the material.py so that I can create a refill file from zero. 

Another problem is, when I use the "REFILL" command for the stratasys-cartrige.py the material will be changed and I don't know why.

Attached there are these files:
400ccm.bin - Dump of the original EEPROM with 24.xx cuin
full.bin - Dump of the original EEPROM with 100%
refill400.bin - created with the refill-command from the 400ccm file
refillfull.bin - created with the refill-command from the 100% file

Silvood, 2017-02-21:
the name showing for the HP ABS Ivory in CatalystEX is "P430L_IVR"
so, I suggest adding a new line based on the P430_IVR, like: id_to_name[0xXX] = "P430L_IVR"

The only thing needed is the hex code for this one. Basically, we just need to find it in the decrypted dump.

256FF, 2017-02-21:
And where can this be found?

Silvood, 2017-02-21:
I think using the hex code stratasys-cli is giving could be good. As seen in issue #21 , 0xFFF (material??) and 0x1FF (support??) I'll try python on my Pi this evening on my HP material dump if I have the time.
I think this could solve the first problem.

For the second issue, I supposed (I didn't even try nor seen the code yet) this is related to the unknown material not supported for refill?
Maybe @bvanheu can help us on this ?

bvanheu, 2017-02-21:
Hi,

It's not the first time I hear about this material id (0xfff) but I'm not sure if it's a valid material? Maybe it's a sentinel to let the printer know this is a special kind of material and it should be found elsewhere in the EEPROM dump?

If you are sure it's the correct material id, then yes you simply need to add a line into this:
https://github.com/bvanheu/stratasys/blob/master/stratasys/material.py

    id_to_name[0x387] = "ULT_S_2"
    id_to_name[0xfff] = "MATERIAL_NAME"

256FF, 2017-02-21:
Afaik adding 0xfff does not work. I even tried it. At the next time reading the dump with this informationaterial is show as unknown 0xffe

256FF, 2017-05-29:
Late, for me too late there is a solution.

For the HP machine silvood found the material codes:
```
new Material() {Id=0xd3,Name="SR30L"},
new Material() {Id=0xdd,Name="P430L_IVR"},
```