bvanheu/stratatools #43: No EEPROM data

ssurya300, 2017-03-04:
Trying to refill a 450mc Chip was able to read and the id is b3-xxxxxxx but it doesn't have eeprom in it it has rw , Is there a way refill the chip? 

I am using Rasp pi 3

HaveBlueXST, 2017-03-05:
450mc chips use SHA-1 protection, so you will need to know the HMAC secret in order to write back to it with a RasPi.

ssurya300, 2017-03-06:
Is there any possibility that the fortus 250cc Chip working with 450mc ? Any Ideas ?

Spider83, 2018-12-04:
> Is there any possibility that the fortus 250cc Chip working with 450mc ? Any Ideas ?

maybe if you have Fortus 250 tri to swap eeprom reader with 450 if they have same connection. but the main problem is SHA-1 protection.

H2Powerman, 2018-12-04:
With an fpga you could brute force this similair to bitcoin hashing and get
the encryption key pretty quick.