bvanheu/stratatools #48: Refilling Uprint

ridingg, 2017-05-12:
Hi, Just wondering if anyone can help me.... I'm currently trying to refill my Uprint support chip using the " bvanheu's Stratasys cartridge EEPROM generator " but each time i create a new "refill.bin " file as showing below, and upload it to the chip, i notice that the UID doesn't change and i get an " S1 Invalid error " displayed on the printer.

Here is the original UID for the support: b3d3bc06d014106d

And here is the command I've used to create the new bin file: 

stratasys-cli.py eeprom --machine-type uprint --eeprom-uid 6d1014d006bcd3b3 --serial-number
    111222333.0 --material-name ABS_S --manufacturing-lot 7567 --manufacturing-date "2016-01-01
    01:01:01" --use-date "2016-01-01 01:01:01" --initial-material 40.0 --current-material 40.0 --key-
    fragment 55aa55c395bf4ebe --version 1 --signature NOTSTRATA -o refill.bin

Please see attached image which shows the original support UID and code for its remaining  4% support, followed by the new code from the refill.bin file.... both shows the same UID.

[image]

HaveBlueXST, 2017-05-12:
The EEPROM UID is read-only - it's actually lasered onto the EEPROM die at the factory.  You only need to change the serial number when generating a new EEPROM image to avoid having to clear out the system.dat file(s).

ridingg, 2017-05-12:
Im gettings an " S1 carrier Invaild " after copying the new code to the chip, but if i copy the original 4% code back it works again with out any errors. any idea how to fix this ?
[image]

HaveBlueXST, 2017-05-12:
I recommend changing only the bare minimum when writing new data to the EEPROM - I'm not sure if using a different signature could be interfering.

ridingg, 2017-05-12:
Tbh im pretty new to this.... All i changed was Rev the UID and entered ABS_S for material as shown above in the earlier screenshot.  I really dont know what else to do

HaveBlueXST, 2017-05-12:
Have you used akb's instructions from this thread?  https://www.3dprintforums.com/showthread.php/3153-How2-Refill-the-EEPROM-of-the-HP-DesignJet-3D-aka-uPrint

ridingg, 2017-05-12:
Yea I've looked at that one, also the Uprint v2  but still getting the invalid carrier error.  We can copy the model code to the chip and it works fine as a model material but for some reason, the support has this error.

HaveBlueXST, 2017-05-13:
Can you post the original support EEPROM dump?

256FF, 2017-05-16:
Hey ridingg, 
AFAIK you need to choose ABS_SS instead of ABS_S to run with the uPrint.

256FF, 2017-05-16:
And voila, after decrypting your 4% dump:

[image]

ridingg, 2017-05-16:
Thanks 256FF, I will give this a try, and let you know if it works :-) ..... Also is there any way of changing the temperature of a Uprint to be able to use other materials?

256FF, 2017-05-16:
I only know two ways to change the tip-temp:

1. maracaEx
[image]

2. System file
I don't know where exactly

ridingg, 2017-05-16:
Yea i've try changing the temp using MaracaEX, But i get an error just before its about to print then reboots itself. 
[image]

256FF, 2017-05-26:
This Code means: "Abort : Head temperature setpoint too low."

Verify the head temperature set point using CatalystEX
• The model must be above 240 degrees.
• The support must be above 240 degrees.

ridingg, 2017-05-27:
Thanks 256FF :-)

You know once i have installed Python, PyCrypto and PySerial and downloaded the bvanheus script. how do i go about making a new code which will work in my uprint for both model and support? Do i just take the UID : b3 d3 bc 06 d0 14 10 6d and input the following information : 

stratasys-cli.py eeprom --machine-type uprint --eeprom-uid b3d3bc06d014106d --serial-number
111222333.0 --material-name ABS_S --manufacturing-lot 7567 --manufacturing-date "2016-01-01
01:01:01" --use-date "2016-01-01 01:01:01" --initial-material 40.0 --current-material 40.0 --key-
fragment 55aa55c395bf4ebe --version 1 --signature STRATASYS -o refill.bin

And change the serial number end number up one, or am i doing this all wrong? Any chance you can explain the process please?

ridingg, 2017-05-27:
Hi 256FF

Any idea why im getting this error when i select ABS_SS to create a support eeprom? 
[image]

256FF, 2017-05-27:
If you use the newest code you must use P400SR instead of ABS_SS.

ridingg, 2017-05-27:
Thanks so much 256FF for all your help...its all working now :-)

ridingg, 2017-05-30:
Hey 256FF, is there a way of increasing the extrusion rate for slightly thinner filaments which are not quite 1.75mm but more 1.74 / 3 you know of ?

256FF, 2017-06-01:
Hi ridingg, 
sorry, I don't know an option to change the extrusion rate. 
But you can try to decrease the distance in insight, also negative values are possible. Try start with -0.0254

[image]

ridingg, 2017-06-01:
Thanks, I will give it a try :-)

I've been trying to use HIPS filament as a support material, which works great, but the hips keep's coming away from the build surface during the print. So I thought I would try changing the first layer of support to model material using the follow changes in the " mvcplr2.cfg" in the hana folder 

FirstVmax=2.0;                  in/sec
FirstVbb=1.5;                   in/sec
FirstVmin=1.1;                  in/sec
FirstBetaVmin=52;               degrees
FirstBetaVbb=35;                degrees
FirstBetaLen=0.1;               in
FirstTempOffset=100;            1/10 C
FirstMaterial=MODEL;
FirstMainFlowAdj=100,1600000,1.75,0;

But for some reason, it still prints the first layer using the support material and not the model material. Is there anything else I need to change to get this to work that you know of?

256FF, 2017-06-01:
Does it warp? 
You can try to increase the base oversize oder base hight to avoid the warping.
[image]
[image]

ridingg, 2017-06-01:
the hips doesn't stick well to the build platform, so i was trying to print the first layer of the support in model material, like in BST

256FF, 2017-06-01:
ah, ok. 
Did you try to use a glue stick to prepare the base? Maybe this can help.

calvindomenico, 2017-12-06:
can anyone point me to the instructions for writing to a uprint chip via the diag port. most of the links I have found are dead ends at this point. 

I am trying to figure out if it is possible to make new chips from scratch. I have done this for fortus/dimension chips

Thank You 
Calvin

256FF, 2017-12-06:
Do you have send me an PM in 3dprintforums.com? If yes, you alreade have it :-)

Belencsakl, 2018-02-15:
Hello Gary Ridingg, 
Did you found the solution for changing the Firstmaterial from support to Modell?
I want same on my dimension.
Please tell me the solution if you found.
Thank you!

ridingg, 2018-02-15:
Hi Belencsaki,
You can change the way the support is placed using the Insight software under "support settings"
Hope this helps :-)
Gary

Belencsakl, 2018-02-15:
Hello Gary,
thanks for your very fast answer.
Ah, yes,  I have other problem:( I want print without any support material in my SST.
I tried in the insight. I made the CMB file without support. I want print the item without support, because not necessary for my parts. But If Build the CMB, my Printer ( Dimension SST768) always make the first layer from support. 
I changed in the CFG file the " FirstMaterial=MODEL; " but doesn't work:( But maybe this is only change the materials sequence:(
Laszlo

ridingg, 2018-03-14:
Hi Belencsald

I have found a way to print a layer of the model onto the build surface before printing the support, which will give the print a very strong layer to stop models from warping.

If you load your model up in the " Insight " software, then under the " Modelers / Setup " tab, you will need to change " Invert build materials " option to "Yes " this will swop the model material and support materials around. You will then have to unload both model and support material from your material, and load your support material, in the model bay, and model material in the support bay.

Once you have sliced your model and sent it to the printer to print, you will notice that it will now print the first layer in Model material from the support side of the print head, followed by the support material from the model side of the print head.

This has worked well for me, without having to ssh and change any settings on the machine.

Let me know how this goes for you and let me know your outcome.
Best wishes
Gary. R

Belencsakl, 2018-03-14:
Re: [bvanheu/stratatools] Refilling a Uprint ? (#48)

Hello Gary,

Many thanks for your respond. 
I'll try tomorrow. I'm very excited:)

I found also 1 "poor" solution:

1, I use a modified plastic blase plate with steel plate + magnet rubber sheet for the calibration before start the print. I attached a pictures
2, I slice the model in insight with 2 layer base from support material
3, a push pause in inside to 3. layer.  ( will stop the print before start the 3. layer from model material)
4, I removed the "support material from the head
5, Start the printing the first 2 layer in to AIR 
6, Printer paused 
7, I push another magnet sheet with "Buildtak". Buildtak heigt is about 0.25mm so to layer hight if I print with 0.127mm
8, And resume the printing.

Because the rubber magnet sheet is flexible. Remove the parts is very easy.

But your solution looks better.

I'll let you know.
Regards
Laszlo from Hungary

Wednesday, March 14, 2018, 1:59:33 PM, you wrote:

Hi Belencsald
I have found a way to print a layer of the model onto the build surface before printing the support, which will give the print a very strong layer to stop models from warping.
If you load your model up in the " Insight " software, then under the " Modelers / Setup " tab, you will need to change " Invert build materials " option to "Yes " this will swop the model material and support materials around. You will then have to unload both model and support material from your material, and load your support material, in the model bay, and model material in the support bay.
Once you have sliced your model and sent it to the printer to print, you will notice that it will now print the first layer in Model material from the support side of the print head, followed by the support material from the model side of the print head.
This has worked well for me, without having to ssh and change any settings on the machine.
Let me know how this goes for you and let me know your outcome.
Best wishes
Gary. R
—

Best regards,
 Belencsakl                            mailto:belencsakl@gmail.com

	
        
		Virus-free. www.avg.com

ridingg, 2018-03-14:
[image]
[image]

ridingg, 2018-03-14:
[image]
[image]
[image]
[image]

TheGuardDawg, 2020-03-23:
Would anyone be able to send me model and support chips? Mine got locked unfortunately... I would greatly appreciate it! Thank you