bvanheu/stratatools #51: Trouble With Rasp Pi UID Reading

swkmmm, 2017-06-27:
We have our Raspberry Pi set up like the example on the README.md. We go to enter the following commands:

HaveBlueXST, 2017-06-27:
It sounds like you may have DeviceTree enabled, as that's the default on newer releases of Raspbian.

swkmmm, 2017-06-27:
So we got to the point of being able to copy the eeprom to the local file. What tool do I use to decrypt it?

HaveBlueXST, 2017-06-27:
Use stratasys-cli.py, just like the readme.md file.

swkmmm, 2017-06-28:
$ ./stratasys-cli.py eeprom -t fox -e 6b0000014d476223 -i cartridge_dump.bin

We attempted to follow the above instructions but got an error. 

ImportError: No module name Crypto.Cypher

256FF, 2017-06-28:
You can find a detailed instruction here: [http://haveblue.org/?p=1988](url)

Toblivar, 2017-07-10:
Make sure to install the pycrypto module, as it is dependent on that library.

bvanheu, 2019-06-30:
Closing this issue as this is not a bug with the actual tool but support request.

closed 2019-06-30 by bvanheu