bvanheu/stratatools #52: Troubles with uPrintSE

lucy707, 2017-07-05:
Hi!
I am having troubles with the uPrintSE chips. I have two main issues.
1. I can't decrypt and print out the information on the chip. I read out the chip via putty and tried to encrypt the information on it and it would give me an error (see image).
[image]

BUT, when I do the same thing with an uPrint chip (not SE), it works perfectly fine and it will print out the information (see image below)
[image]

So, I kinda had the feeling that this is an uPrint SE specific issue. I'm confused why it works fine with an uPrint chip but not with an SE. Anyone any ideas?

2. Issue: If I overwrite my uPrint SE chip, the printer will not recognize the material. It says "invalid carrier".
I used this to generate the new hex codes and then copied it on the chip.
`./stratasys-cli.py eeprom --machine-type uprintse --eeprom-uid b61014d006cec1b3 --serial-number 7070.0 --material-name P430_IVR --manufacturing-lot 7070 --manufacturing-date "2001-01-01 01:01:01" --use-date "2002-02-02 02:02:02" --initial-material 40.0 --current-material 40.0 --key-fragment 4141414141414141 --version 1 --signature M2 -o three_percent.bin`

Here is the UID and the information on chip:
b3 c1 ce 06 d0 14 10 b6
e7 6d 89 74 99 43 53 72 04 50 3d 84 c4 27 2b 62 5b e3 65 36 86 53 e3 c8 10 28 38 f4 45 d5 3f 0c f1 35 ac 94 bd bc c8 12 cf 48 39 87 fa 0a 6e 25 cf 48 39 87 fa 0a 6e 25 80 b8 88 20 b5 95 72 87 42 c4 00 00 00 00 c6 53 55 aa 55 47 9e e8 4e be 6d 0f 00 00 00 00 00 00 c8 3a 9a 08 26 d2 ea 84 8f bc 9d c9 01 00 00 00 53 54 52 41 54 41 53 59 53 02 34 f1 de 56 45 bd 32 46 60 31 21 a5 c8 fa

256FF, 2017-07-11:
Please try to use "--signatrue STRATASYS" for the creatoin of the new chip. AFAIK I've had issues when using others. 

Reading the SE: Same failure here. Maybe the ID of the SE (6B2A268B5ED3374A) is not correct. When you open the https://github.com/bvanheu/stratasys/blob/master/stratasys/machine.py you can see that it is the same as "ktype" (6B2A268B5ED3374A)

cdnreprap, 2017-09-14:
Did this ever get answered?

I'm having the same issue.

256FF, 2017-09-14:
Yes, machine ID is the same for all uprint models. Please choose uprint for decrypt

cdnreprap, 2017-09-14:
have you actually confirmed this?

256FF, 2017-09-15:
sorrx, I think a postet a wront info. The IDs are probably not the same.
I can't read a SE dump with the uprint machine ID. And the SE ID seems to be wrong.

bvanheu, 2017-09-15:
So this machine id '6B2A268B5ED3374A' works for Uprint but not Uprint/SE?

I can update the code to reflect this.

256FF, 2017-09-15:
No, the ID F3A91DBE6B0B2255 works for uprint(+) and Hp DesignJet.
The ID for SE(+) is not confirmed, AFAIK.

256FF, 2017-09-15:
Another thin bvanheu, 
how do you have found out the machine-IDs of the other printers?

bvanheu, 2017-09-15:
256FF: they were given to me by email.

See this commit: 71320cdbdeeb08009371f5ccfebe5af85ba5d7a2

thekakester, 2017-10-06:
Has this issue been addressed?  I also have an SE(+), but I'm unable to decrypt the dump.
Is there anything I can do on my end to help solve this?  I could send dump files or mail physical cartridges if that makes anything easier.

thekakester, 2017-10-06:
Does the EEPROM UID play into the encryption?
To start, I'd like to see if I can "clone" one of my full chips to an empty one.  I know the printer keeps track of serial numbers, but I'd like to see if I can write SOMETHING to my cartridge that the printer will recognize.

Right now, I'm using @256FF 's cartridge writer (https://github.com/256FF/CartridgeWriter) and I commented out all the safety checks (like checksum validation) to see if I could write the same encrypted data to a new chip.  Currently, this does not seem to work, but I'll post updates as I go along.  Any tips or suggestions would be greatly appreciated :)

Silvood, 2017-11-28:
@thekakester I'll happy to help you in this task. My last uPrint chip have gone ReadOnly (forgot to refill it before printing for a customer yesterday....) And I have only uPrintSE chips available that I can't use because I can't rewrite them :/

So basically, my printer is useless right now and I need it ^^ I'll do anything including this bruteforce to advance on this SE machine code

256FF, 2017-11-28:
Hey silvood,
Afaik you have my email, right? Contact me and I can send you one or two uprint chips.

Silvood, 2017-11-28:
@256FF , thanks for the offer, but I was able to find one with a friend (stratasys resseller), he asked one of his customer to give me one ^^

Nordeal, 2018-07-13:
Hello everyone,
I'm also interested in re-using cartridges for uprint se plus so I'll share my knowledge and maybe you'll help me in further experiments. So what everybody already knows is that uprint se uses DS2432 EEPROM that is SHA-1 encrypted and one need NDA with MaximIntegrated to get information about decrypting it. But what I did managed to do is to make a backup of EEPROM through UART interface of the printer and also successfully restore it to the same EEPROM but after some changes(printed 1% out of it). Though 3d printer recognized it as corrupted after I restored previous backup. After I restored actual backup printer recognized it again, which means that uprintse saves eeproms he worked with on HDD too. So the next logical step is to get to HDD, create backup of it and try to find where it keeps data about eeproms. The problem is that I haven't found any videos or manuals how to open uprintse printer and where to search for HDD. Does anyone have teardown experience?

HaveBlueXST, 2018-07-13:
Nordeal, the hard drive is in the rear electronics bay.  Remove the top panel (3 screws on the top rear), then the side panels (3 screws each, also on the rear), remove the filament tubes from the Y-connectors, then remove the 3 screws at the top of the electronics bay (bottom panel on the rear).  The electronics bay door should swing down, and you should be able to see the drive in the upper left of the bay.

I believe the EEPROM data is stored in a file called system.dat, and I seem to recall that it keeps a second copy somewhere else on the system that also needs to be cleared out if you want to re-use EEPROMs.

Nordeal, 2018-07-19:
HaveBlueXST, thank you for quick answer. With your help I managed to find HDD, clone it and make some changes. I removed system.dat and set rc.local to remove this file on each start. Saved eeprom state, printed 1%, returned saved state, restarted printer and it worked! The printer said he has 1% more material as before restart. So I've left it to print a big part overnight. In the morning there was 38% material left. Backed up current eeprom state, written previous, restarted printer and on the restart the printer did something he never did before - unloaded material and after that it said that material bay is empty, eeprom read-only, couldn't even return it to previous state, I also tried to get a bit more information from printer using logs, but it didn't create log files neither in /var/log nor in /etc. Attempt to start sshd also failed, though I added "/etc/init.d/sshd start" to rc.local and access to ssh port in iptables

altug19, 2019-05-28:
Hi,

I have Uprint SE. In our country their cartridges are too expensive and i'm looking for eeprom hack. Can you help me how to do it ? I want to hack eeprom chips and i don't know how ?

256FF, 2019-05-28:
Hi,

There are two options for you

1. Edit the Linux of your printer to delete the cache files where the eeprom ID's are stored at startup. Dump the full eeprom of your chips. Make sure you don't print lower than 2%. Rewrite the eeprom and reboot. The chip is full again.

2. Convert the printer to a non SE one. Than there is a program from silvood to flash the eeprom through rs232 with a new serial number so that you don't need to clear the flash file.

Kind regards

Von: altug19
Gesendet: Dienstag, 28. Mai, 11:36
Betreff: Re: [bvanheu/stratatools] Troubles with uPrintSE (#52)
An: bvanheu/stratatools
Cc: Odin, Mention

Hi,
I have Uprint SE. In our country their cartridges are too expensive and i'm looking for eeprom hack. Can you help me how to do it ? I want to hack eeprom chips and i don't know how ?

kamada-korg, 2019-06-27:
Hi, I have same issue as @altug19 , and I have a question.

I have uprint SE, not uprint.
If I could get the uprint chip, I can rewrite the uprint chip and use the chip for uprint SE?

Best regards,

bvanheu, 2019-06-30:
I'll close this bug because the following ID can be used for UprintSE `09FBD4B61FC0B327`.

See https://github.com/bvanheu/stratatools/commit/41accb5cc35dfcbb94298694bc5f60aaaa82307c

closed 2019-06-30 by bvanheu

TheGuardDawg, 2020-03-21:
> Hey silvood, Afaik you have my email, right? Contact me and I can send you one or two uprint chips.
> […](#)

Hi Oden,

My uprint chips also went read only... would you be able to help? :) Not sure what to do.

altug19, 2021-03-03:
hi, I have Uprint SE andi used cartridge writer program from riverfrontam. I used on model filament and worked well. Then , my support material was at %1 percent and i used read button and then pushed decrypt button and filled the support material but i forgot to change serial number part. Then after  i load the support printer said cartridge is empty. I think machine has logs for serial numbers and didn't accept the %100 support material after changing. Do you have any suggestions for me ?

0815isnt, 2022-02-01:
> hi, I have Uprint SE andi used cartridge writer program from riverfrontam. I used on model filament and worked well. Then , my support material was at %1 percent and i used read button and then [...]

Did you solved your problem? If not, please send me your crypted data and I will send the new data back to you.

altug19, 2022-02-01:
> Did you solved your problem? If not, please send me your crypted data and I will send the new data back to you.

**hi, i faced the same problem. Machine logs the serial number for sure. Please watch this video and follow the instructions; https://www.youtube.com/watch?v=_YLUCKgAaJk&ab_channel=ScrapTechTips.
you should back up the cartridge filament datas before the procedure. you have two possibilities in front of you. One is eeprom on your support filament is on read only mode or not. I have my old cartridge data for the support material for back up purpose. I can send the datas for you and you can write the data on your own eeprom. You see the procedure on the video how you should do that. The datas on my other computer please write your email here and i will send the datas for your testing.      **

altug19, 2022-02-03:
> Did you solved your problem? If not, please send me your crypted data and I will send the new data back to you.

yes i solved the problem thank you very much