bvanheu/stratatools #53: Stratasys Fortus 450mc

Toblivar, 2017-07-07:
Has anyone had any luck reading/writing the 450mc chips?  I've noticed they are slightly different in how they store everything (UID is 23-xxxxxxxx, no eeprom file but a "rw" file, "rw" file shows as all f values when viewed with xxd).  I also understand that they use SHA-1 encryption on these.

If anyone can help point me in the right direction, I would appreciate any help available.

Thanks!

fsabbatini89, 2018-03-02:
Helo Toblivar, did you manage to solve this?

Mickele84, 2018-04-12:
News?

cj1294, 2018-04-16:
I'm also in a similar situation with the 250mc

Mickele84, 2018-04-17:
🙏🏼😅

Thanks.

Saluti

Michele Rossetti

Inviato da iPhone

tjschweizer, 2018-06-09:
So I've been poking at this for a few days now and I can't seem to make any headway. I've got a few dozen of the chips for the fortus 450 filament, so if anyone needs more scans let me know.

Attached is an excel file containing two hex dumps from one filament canister. Here is the data I get about the filament from the printer itself:

Name: "ASA_RED"
Color Label "Red"
Color Code "A6192E"
Family Name "ASA"
Family Display Name "ASA"
MFG Date: "02052018" - Feb 5th 2018
MFG Lot: "104582"
Initial Matl - "92.31" (cu.in.)
Serial Number "414253482"

So I got the hex dump from the canister, then unloaded some filament and got the hex dump again. 
Here's the amount:

Before unloading filament, Remaining Material = "3.95126" (cu.in)
After unloading filament, Remaining Material = "3.45126" (cu.in)

Looking at the hex dump, the only difference is in the first 12 bytes of data. It goes from:
B6C0C36D4C996BE81E69C2A - before, to
9308DBDD63796E351DABC06 - after

I'm not even trying to write to this EEPROM, I'm just trying to read the remaining material from the canister without putting the canister in the printer...

Anyways, if anyone can crack this code that would be great, because I am at wit's end.

[414253482.xlsx](https://github.com/bvanheu/stratatools/files/2087433/414253482.xlsx)

fsabbatini89, 2018-06-10:
Hello everyone,
I am making some progress qith this but i had just not enough time to test my solution.. i am not reading and not writing the chip. I noticed that the secret key is in the pic32 located at the canister controller board and the data is transfered raw from the computer board to the canister board through rs422 twisted pair cable and to nake it gull duplex it uses two pairs of twisted cables. I just sniffed the exchanged data and reverse engineer the communicatikn protocol, piece of cake by the way..
I made a small board that i put in me middle of the communication between the pc and the cannister control board so when the amount of remaining material is written to the canister this small board just overrides it and answer for the canniater ( the canister does not even realize the pc tried to communicate with it!!).
I an post the code ig anyone wants to help with it!

tjschweizer, 2018-06-10:
fsabbatini89,

Do you have any pictures of how you accessed the canister controller board?Or did you just use the diagnostic port on the back of the printer?

tjschweizer, 2018-06-10:
Yikes, that's an expensive warranty to void!

Have you tried sniffing the lines while the printer is checking the canister itself? I'll try getting a log of that tomorrow, and maybe see if I can get to that control board. 

When you say erasing the serial number of the cartridge from the PC resets the material, does that mean the material count isn't actually stored on the EEPROM? That's.... disheartening.

Either way, keep me updated and I'll post some data when I get it! Enjoy the beer haha

fsabbatini89, 2018-06-10:
Haha i know! I managed to fix it with 2 dollars but the MIO board costs 4500 dollars and the cannister its about 5000 dollars!
The material count IS stored in the epitome but if the cannister board never gets the order to write into it, it will never get empty!!! So how it works? It is a pcb that you put inbetween the pc and the canister board. The pc always after a write into the cannister it executed a read command based on my discoverings. So if I just detect this write command, store in the microcontroller the string it wants to write into the eeprom and when the pc reads it back (the next command) the eeprom does not answer but it is the microcontroller with the very same data the pc “wrote” into the eeprom, the pc thinks the write command was successfull but the eeprom is like new! 
It is clearer now the workings of this? I just have to test it out but the pcb and the microcontroller with the snuffed communications works like charm. I just had had not enough time to test it in the real machine!!!

tjschweizer, 2018-06-10:
Oh that's clever!

If/when you get a chance, can you post a trace of the full write command? Or just a full serial log between the pc and canister board?

fsabbatini89, 2018-06-10:
Yes, i can do it . I will try it to do it when i get home tonight along with my interpretation of the commands. Anyway no you don’t even have to care about what the write string means, as long as the pc gets the answer it expects it will be happy and continue printing

tjschweizer, 2018-06-10:
Awesome!

I'm not even trying to get this EEPROM info to write to it - I just want to be able to read it. I have maybe 20 canisters that I have to log the remaining material in, and I want to just do it with a microcontroller without having to load the canister into the printer.

Seeing the write message before it is encrypted will go a long way towards figuring it out. Plus if you share that string, then I don't have to break my printer getting to the serial traces :stuck_out_tongue_winking_eye:

tjschweizer, 2018-06-10:
Here's the same set of files but after running through a load-unload cycle (uses roughly 0.5 cu.in. of material)

[EEPROM Dump - Yellow Load_Unload1.log](https://github.com/bvanheu/stratatools/files/2088177/EEPROM.Dump.-.Yellow.Load_Unload1.log)
[ASA Yellow Load_Unload1 TCP Stream.txt](https://github.com/bvanheu/stratatools/files/2088178/ASA.Yellow.Load_Unload1.TCP.Stream.txt)

(Oscilloscope trace incoming)

tjschweizer, 2018-06-10:
And after another load-unload cycle

[EEPROM Dump - Yellow Load_Unload2.log](https://github.com/bvanheu/stratatools/files/2088191/EEPROM.Dump.-.Yellow.Load_Unload2.log)
[ASA Yellow Load_Unload2 TCP Stream.txt](https://github.com/bvanheu/stratatools/files/2088190/ASA.Yellow.Load_Unload2.TCP.Stream.txt)

[Oscilloscope Trace](https://www.dropbox.com/s/klmkkyq7ys7pwp5/ASA%20Yellow%20Load_Unload2%20-%20scope%20sniff.csv?dl=0)

tjschweizer, 2018-06-11:
Here's the OneWire Hex values of the Oscilloscope trace

I'm going through and checking for common commands and stuff

[ASA Yellow Load_Unload1 - scope sniff.txt](https://github.com/bvanheu/stratatools/files/2088635/ASA.Yellow.Load_Unload1.-.scope.sniff.txt)
[ASA Yellow Load_Unload2 - scope sniff.txt](https://github.com/bvanheu/stratatools/files/2088636/ASA.Yellow.Load_Unload2.-.scope.sniff.txt)
[ASA Yellow Full - scope sniff.txt](https://github.com/bvanheu/stratatools/files/2088637/ASA.Yellow.Full.-.scope.sniff.txt)

tjschweizer, 2018-06-12:
@fsabbatini89 Did you get a chance to get a log of the commands?

fsabbatini89, 2018-06-12:
I will try to post them tonight!

fsabbatini89, 2018-06-14:
@tjschweizer  I am sorry i could not do this before but the files where in another pc i have so i wasnt able to find it before!
Attached you will find a raw sniff of the communication PC<->MATERIAL BAY and a file with the interpreted command, CRC checks and everything you should now of the communication. As i said, i dont even care of the amount. I use metric (cm3 i think or maybe mm3, dont remeber! but i know the ull catridge is 1500 something3 hehehe, make the conversion). Because my idea is just cut the communication as soon as the pc sends a write command to the bay and also replace the sequential number from now on (i did not get a time to test my pcb in the machine but it is already tested sending this strings directly to my pcb and listening from the output of the pcb..)

[DUMP_374040012_completo HEX.txt](https://github.com/bvanheu/stratatools/files/2102484/DUMP_374040012_completo.HEX.txt)
[INTERPRETED COMMANDS.txt](https://github.com/bvanheu/stratatools/files/2102485/INTERPRETED.COMMANDS.txt)

tjschweizer, 2018-06-25:
So if I get access to the HDD, does anyone know what exactly I should be looking for?

tjschweizer, 2018-08-07:
@fsabbatini89 

Do you remember what the baud rate of the RS422 is?

fsabbatini89, 2018-08-07:
Yes, it is 115200. Remember that it is a dual RS422 (full duplex), one pair
receives and one pair sends

jerryjonesiwe, 2018-09-27:
Hello,

New to the forum here and want to help. 

What value pull up resistor are you using?

fsabbatini89, 2018-10-18:
Hello everyone, any updates on this matter?

fsabbatini89, 2018-11-03:
@jerryjonesiwe @tjschweizer
Did you make any progress decrypting the file?

Spider83, 2018-12-04:
Hello. any news about this project? I have Vantage Se and Fortus 450mc. Now i`m using raspberry to reflash eeproms on Vantage SE chips. If I can help with testing or smth else - here i am )

fsabbatini89, 2018-12-04:
I think I am almost there. I would need some logs of filled cartridges of
different materials if you have. Fortus 380 or 450

Spider83, 2018-12-04:
> I think I am almost there. I would need some dumps of filled cartridges of different materials if you have. Fortus 380 or 450 El El mar, 4 dic. 2018 a las 02:48, Spider83 <notifications@github.com> escribió:
> -- Franco

if you give me instructions how to make them - i`l send them to you. I have all kind of materials with support for 450mc except St130 and Antero.

fsabbatini89, 2018-12-04:
With the chip reader you have, with the raspberry just read and out it in a
file

Spider83, 2018-12-04:
ok. will try it tomorrow and then write back. Today have waork to do.

Mickele84, 2018-12-04:
Good job!!

Saluti

Michele Rossetti

Inviato da iPhone

> Il giorno 4 dic 2018, alle ore 11:21, fsabbatini89 <notifications@github.com> ha scritto:
> 
> I think I am almost there. I would need some logs of filled cartridges of
> different materials if you have. Fortus 380 or 450

fsabbatini89, 2018-12-06:
@Spider83
What kind of filament do you use for your vantage, brand and type?

fsabbatini89, 2018-12-09:
Check this out
[Stratasys hack](https://www.youtube.com/watch?v=goqdKioSI9o)

fsabbatini89, 2018-12-09:
> I don't get it - this doesn't do anything to modify the EEPROM, right? It
> just erases the temporary value stored on the hard drive before the printer
> writes that value to the EEPROM?

It does not allow the value to get to the eeprom

tjschweizer, 2018-12-09:
Ahh. In that case, I think a simple "sudo service masterd restart" would probably work instead of a full reboot.

Is this your video? Has anyone been able to enable ssh on the 380/450?

Also sorry for not responding to the earlier messages - work got insane and is finally dying down. I am still working on decrypting the EEPROM, but who knows.

I never got my EEPROM dumps from the diag port on the 450 - I got them using an arduino.

fsabbatini89, 2018-12-09:
Anyway if you decrypt the eeprom you will not be able to write it as long
as you    Don’t have the secret

In the video you can see how after printing just unplug and plug the
cartridge gives full amount again

regedit2, 2019-02-23:
Hi, Iam new here. Whats the status? You are able to read and decrypt the eeprom. But you cant write because we havnt the sha secret?

regedit2, 2019-03-15:
I start makeing some EEPROM dumps. If anyone is interested please ask. But I dont want to share them here publicly.

Leitnin, 2019-04-03:
> Check this out
> [Stratasys hack](https://www.youtube.com/watch?v=goqdKioSI9o)

Can you provide some more details about this?

jerryjonesiwe, 2019-04-03:
Has anyone here on the thread managed to reprogram the plus canister EEPROM. I’ve been on this thread for a while but lost touch. 

Jon Thomas

fsabbatini89, 2019-04-04:
@Leitnin
Yes,
Put a new cannister, use it use it and use it. When it is empty just remove the cannister and fill it up. Reboot the machine and voila, the cartridge is full again.
It works as a charm. You have to use a man in the middle pcb in between the cartridge and sbc

Leitnin, 2019-04-04:
@fsabbatini89 

Thank you for the speedy reply. I appreciate it. I'm sure though, you can imagine that while that sounds great and like exactly what I and others are looking for, it really isn't sufficient information for someone to understand what needs to be done or what they need to acquire to make this work.

1. Details on the is the man in the middle pcb? 
2. What about the files that need to be deleted if over 30cm^3?

fsabbatini89, 2019-04-04:
I created a file that gets executed at startup that deletes a file that
holds the amount of a wire in a used catdrige
I can upload a picture tomorrow of the pcb I made

jerryjonesiwe, 2019-04-04:
Awesome. My goal is to be able to take an EEPROM and set to 92. Also if it’s possible to reprogram into another material that’s great. 

I know what it can be done because I have a friend at Stratasys and he says the box is no bigger than a cigarette pack and they reuse all the time. 

Just need to get around encryption. 

Jay

7786, 2019-04-12:
> I created a file that gets executed at startup that deletes a file that holds the amount of a wire in a used catdrige I can upload a picture tomorrow of the pcb I made El El mié, 3 abr. 2019 a las 22:00, Dan Leitner <notifications@github.com> escribió:
> -- Franco

We're also very interested in more information about this PCB if you're willing to share, we'd love to do this to our work printer.

fsabbatini89, 2019-04-12:
Hi!
I am moving my machine to a new location. As soon as i plug it again i will
show you how it works

fsabbatini89, 2019-04-19:
Ok, Man in the middle technique.
In the attached pictures you can take a look how the system is connected. My custom PCB is located in the middle between the cartridge bay pcb and the mainboard. You can see in the pic below that each cartdrige bay has its own man in the middle pcb.. 

The only thing to solve is make a Cron job to delete the file so you can change the cartridge when it is empty without having to turn off and on the machine

[image]
[image]

Mickele84, 2019-04-19:
😲😲👌🏾 good job.

Saluti

Michele Rossetti

Inviato da iPhone

spencermclegg, 2020-01-31:
@fsabbatini89, I'd like to try your hack. Are you willing to provide the details (schematic, etc.)? anyone else been able to ssh into the 450 and work around the filament eeprom check?

fsabbatini89, 2020-02-01:
Hi! I have been using my hack for a long time without any issues!

fsabbatini89, 2020-02-01:
I can provide the schematics and the hex so you can test it in your machine

spencermclegg, 2020-02-03:
@fsabbatini89, yes, send the schematics and source my way. spencermclegg [at] gmail

Spider83, 2020-02-04:
> -- Franco

can you send it to me too? kruchok [at] gmail

7786, 2020-02-04:
> -- Franco

Hey Franco, I would love to try out this method too if you don't mind, it would be much appreciated. prime4prime77 at gmail

Basement-Modder, 2020-02-06:
> -- Franco

I would also appreciate an email. I have been toying with these new chips for a while and haven't really gotten anywhere. bill.the.williamson at gmail

fsabbatini89, 2020-02-06:
Hi everyone, sorry for the delay, I am on a trip right now and I will be
back by the end of next week. I will try to do my best to send you the
schematics and the soft. The cob is based on a dspic with 2 uarts (serial
ports) and a four rs485 driver chip from maxim. Are you able to buy this
stuff from mouser or digikey? Also the dsp only comes in surface mount
package soic for easy welding, are you able to send the pcb to a fab??

fsabbatini89, 2020-02-06:
I almost forget, you also need a pickit 3 to send the hex to the device!

spencermclegg, 2020-02-06:
pickit 3/4, surface mount parts, and the rs4585 parts are not a problem. Looking forward to the schematic and source code.

Puddle-Jumper, 2020-02-07:
Same here. Can you send it to me too? tempmail348 at abwesend de
Whats with the cronjob?

fsabbatini89, 2020-02-07:
The cron job has to be improved so it runs sporadically.. I just made a
script that runs only at startup and erases a file that holds the used
serial number of the cartridges but it can be run sporadically so you don’t
have to turn off and on the machine after changing spools with same number

spencermclegg, 2020-02-19:
@fsabbatini89 still waiting for the schematic and source.

[26 more replies]