bvanheu/stratatools #55: UPrintSE checksum issue

ericalbers, 2017-10-28:
I'm using the github/meawoppl/eepromTool-ds2433.git repo to read the uprint se chips and it seems to get a good read of 512 bytes (the code read/verifies 5 times)
Yet when I call the stratasys-cli program it give a checksum error.
I've tried 3 different chip reads and all give the checksum issue.
Please see attached screenshot and hex dump
Thanks!
[image]

HyKECh, 2018-01-27:
uPrint SE use different machineID - 6B2A268B5ED3374A

256FF, 2018-03-05:
Can anyone confirm, that the machine ID above (fox2) is suitable for uPrint SE too?

bvanheu, 2019-06-28:
@256FF try this one: `09FBD4B61FC0B327` ( https://github.com/bvanheu/stratatools/commit/41accb5cc35dfcbb94298694bc5f60aaaa82307c )