bvanheu/stratatools #65: uPrint EEPROM uid starts with 00 and keeps changing

tvsamuel444, 2018-03-03:
I'm trying to use this tool with a uprint eeprom using the one pin data method on a raspberry pi. The uid seems to keep changing every couple of seconds and it starts with 00- rather than the 23- that the guide says it should have. Anyone have any insight on this problem?

256FF, 2018-03-05:
uprint is a bit special. You are definitly unable to write to this chip. It is possible to decrypt the data with bvanheus script but I have not had success to read it with this script. I found a script (tool.py) do dump the uprint chip and drcrypt it with bvanheus script. Take a look at my github and at @mayrthom. There is help for uprint .... to read, decrypt, recode and re-write it.

tvsamuel444, 2018-03-06:
Im currently waiting for a null modem cable to read and write with the daig port to arrive. I'll definitely try that when I get it. Where is this tool.py script?

mayrthom, 2018-03-07:
You can find the several scripts here: https://github.com/256FF/Refill-uPrint-EEPROMs
For communication over the DIAG Port you can use any Serial Terminal program.

I think it may be easier with my rewrite of the cartridgewriter, because it is an all in one solution for the Stratasys uPrint. It can read, decrypt and rewrite the Cartridges over the DIAG Port in one go.
You can download it here: https://github.com/mayrthom/CartridgeWriter_uPrint

If you have any questions feel free to ask.

256FF, 2018-03-09:
Someone worte, that Machine ID for SE is 6B2A268B5ED3374A (same as "Ktype") but it was not confirmed yet. Furthermore mayrthom also has this ID in his code (machine.cs) but outcommented:

//new Machine {Number = new byte[8] {0x6B, 0x2A, 0x26, 0x8B, 0x5E, 0xD3, 0x37, 0x4A}, Type = "uprintse"}

cdnreprap, 2018-03-28:
has anyone located the correct machine ID for the uprint SE yet?

bvanheu, 2019-06-30:
@cdnreprap see this commit https://github.com/bvanheu/stratatools/commit/41accb5cc35dfcbb94298694bc5f60aaaa82307c