bvanheu/stratatools #66: EEPROM ISSUES!

williamhmic, 2018-05-07:
Hello. I'll get straight to the point; I've got a Raspberry Pi, have a eeprom hooked up, but can't get the program to recognize the chip. I tried to find a solution but was unsuccessful, I've been using a guide on haveblue.org which I only have the html file for, no images or anything (School blocked it due to being a "weapon") and I've gotten a good bit done. I believe I've gotten up to the section of device tree. After that though, kind of just fell apart. Anyway, I honestly don't know how I should do this. Not sure how to ask this... Could someone help me out? I'm not sure what I should ask, and I've honestly have no idea what I'm doing... I should have done more research, but it IS near the end of the school year. Also, sorry, this is the best place I know to ask for help. not sure if I should ask for help here or else where... Thank you, sorry if I'm confusing, Please help... https://haveblue.org/?p=1988
I have a Raspberry Pi B+; Adafruit Raspberry pi B (Not B+) that I've got working; gator chips hooked up and a dimension 3D printer. Sorry, please help, thank you.

williamhmic, 2018-05-21:
Update: I have been able to find the UID I believe. 23f6138900000010 is what is printed. However, going to the w1 bus master, 1 for me it is w1_bus_master as opposed to w1_busmaster, 2 xxd -p eeprom does not work, instead I need to go to another location 23x0000008913f6/eeprom which returns the ton of numbers as seen in yours. Now, my new issue is, what do I do with the numbers? As far as I can tell, the UID was stareing me in the face. 23x0000008913f6 but reverse every 2 digits... Or.. Well, 00 10 20 30 would be 30 20 10 00. Anyway, next, I unpacked the zip, but I still don't see a stratasys-cli.py in the exctracted to location. I ran the setup.py and that did do SOMETHING. But still no cli.py... I'm confused.. I also should be making notes.. I had something that I just needed to get the uid for. I forgot all about it. I will continue looking for it, but thought I'd ask incase I can't find it again or I do find it, and it does actually need the cli or something. Anyway, thanks! :D @HaveBlueXST :D (Do @'s work?)

williamhmic, 2018-05-21:
What I have: 
build  LICENSE   README.md  stratatools
dist   protobuf  setup.py   stratatools.egg-info

HaveBlueXST, 2018-05-21:
Well, that guide is over 2 years old at this point, and Benjamin has made a lot of updates to the software in that time (not to mention changes in Raspbian itself since then).  Make sure to read through the current readme for information on current usage, since stratasys-cli doesn't exist in the current version.

williamhmic, 2018-05-21:
YOU SHOULD DO A RE-RE VISIT! Heh. Thank you for your help!

HaveBlueXST, 2018-05-21:
Funnily enough, I noticed the 'wiki' section in github projects, and figured that would be an excellent spot to put an updated tutorial.  Unfortunately, my free time as of late is pretty much zero, so getting around to it may take quite some time.

HaveBlueXST, 2018-05-29:
I don't - do you get an error when trying to get the crypto package?

williamhmic, 2018-05-29:
Ok, trying sudo get crypto, it said I have the most recent version.

And thus, I'll post what I get when I put the following command in.

stratatools eeprom_decode -t prodigy -e 1000000098316f32 cartridge_dump.bin

Traceback (most recent call last):
  File "/usr/local/bin/stratatools", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2876, in <module>
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 451, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 464, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 644, in resolve
    raise VersionConflict(dist, req)
pkg_resources.VersionConflict: (six 1.8.0 (/usr/lib/python2.7/dist-packages), Requirement.parse('six>=1.9'))

williamhmic, 2018-05-29:
From the last line, I'd guess something is not meant to be used with a certain version of something else...

williamhmic, 2018-05-29:
Note the bolded requires. That's 2 underscores on either side...

bvanheu, 2018-05-29:
It looks like your package of "six" is too old (1.6.0) and you need >= 1.8.0.

Can you try:
```pip install --upgrade six```
or
```pip install six=1.8.0```

williamhmic, 2018-05-29:
OH OH! Ok, now I've got no file/directory of cartridge_dumb.bin... BUT ATLEAST IT'S FARTHER! YAY YAY YAY.

What do I do about cartridge dump?

williamhmic, 2018-05-29:
I tried 
`stratatools eeprom_decode -t prodigy -e 1000000098316f32 /home/pi/stratatools-master/cartridge_dump.bin`
but that errored... I did make a blank file named cartridge_dump.bin

williamhmic, 2018-05-29:
`Traceback (most recent call last):
  File "/usr/local/bin/stratatools", line 9, in <module>
    load_entry_point('stratatools==3.0', 'console_scripts', 'stratatools')()
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 217, in main
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 31, in run
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 131, in command_eeprom_decode
  File "build/bdist.linux-armv7l/egg/stratatools/manager.py", line 53, in decode
  File "build/bdist.linux-armv7l/egg/stratatools/manager.py", line 209, in decrypt
  File "build/bdist.linux-armv7l/egg/stratatools/manager.py", line 231, in build_key
IndexError: bytearray index out of range`
That was the error... bytearray out of range... What... Please tell me it's just because the file is wrong or something... Just a simple command issue, wrote something wrong... I FEEL SO CLOSE! This is my biggest project so far.. Thanks for the help already everyone! :D

bvanheu, 2018-05-29:
`cartridge_dump.bin` should contain the encrypted cartridge data (the data contained on the eeprom itself). 
You can also point the command directly to the EEPROM, ie:
```
stratatools eeprom_decode -t prodigy -e EEPROM_UID_HERE /sys/bus/w1/devices/w1_bus_master1/23-xxxxxxxxxxxx/eeprom
```

williamhmic, 2018-05-30:
Thanks so much! :D However, now I'm getting a checksum. 
`Exception: invalid content checksum: should have 0xe9ed but have 0x7f92`
Would that be due to wrong printer being selected? Dimension is prodigy right?

williamhmic, 2018-05-30:
Do I have to provide everything for the create? Or can I just do `create ----current-material 22.2`
With only the current material parameter. No seriel. Or will I have to program something that gets the info, only changes the current mat. and then creates a new?

bvanheu, 2018-05-30:
Yes you need to be `root` to write to the EEPROM file (so use `sudo`!)

williamhmic, 2018-05-30:
Ok, I got it! reading aver encoding gives this:
`serial_number: 45988607.0
material_name: "ABS"
manufacturing_lot: "EMPTY"
manufacturing_date {
  seconds: 1183649899
}
last_use_date {
  seconds: 1183649899
}
initial_material_quantity: 56.3
current_material_quantity: 56.3
key_fragment: "55aa550728127abe"
version: 1
signature: "STRATASYS"
`
But the machine says it's empty. Why? I believe it might be manufaction lot being empty. It checks that and then it checks material. Or am I mistaken?

williamhmic, 2018-05-30:
So there's still an issue

bvanheu, 2018-05-30:
you need to update those fields:
```
    cartridge.current_material_quantity = cartridge.initial_material_quantity
    cartridge.last_use_date.FromDatetime(datetime.datetime.now())
    cartridge.manufacturing_date.FromDatetime(datetime.datetime.now())
    cartridge.serial_number = get_random_serialnumber()
```

williamhmic, 2018-05-30:
where should I run these?

williamhmic, 2018-05-30:
Serial number can be random? Does it set the chip's serial number as empty on the printer? Why must one redo the date? I want to be able to edit the txt file I convert to the eeprom file directly.

UsernameWasTaken, 2018-05-31:
I'm stuck at a similar point, I was able to create the image using the commands on the README page. However I don't know what I am ment todo after that. I tried haveblue's 'cp' command to write the file however I fear that all that has done is corrupted the eeprom. Anyway I am unable to decode the eeprom to test of it worked. I don't know if anyone has got any suggestions on what todo next.

HaveBlueXST, 2018-06-01:
Sounds like it could be an electrical/connection problem at that point?  I'd suggest creating a .hex file full of 0x012345678ABCDEF type of patterns and writing/reading it to the EEPROM to make sure that reads/writes are working as intended.

williamhmic, 2018-09-05:
Still no... :/
```stratatools eeprom_decode -t prodigy -e a7027800000023 /sys/bus/w1/devices/w1_bus_master1/23-0000007802a7/eeprom >/home/pi/Desktop/cartdump.txt
Traceback (most recent call last):
  File "/usr/local/bin/stratatools", line 9, in <module>
    load_entry_point('stratatools==3.0', 'console_scripts', 'stratatools')()
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 217, in main
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 31, in run
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 131, in command_eeprom_decode
  File "build/bdist.linux-armv7l/egg/stratatools/manager.py", line 54, in decode
  File "build/bdist.linux-armv7l/egg/stratatools/manager.py", line 112, in unpack
Exception: invalid content checksum: should have 0xdd47 but have 0xd7d2
```

williamhmic, 2018-09-05:
Still no :/
```
stratatools eeprom_decode -t prodigy -e a7027800000023 /sys/bus/w1/devices/w1_bus_master1/23-0000007802a7/eeprom >/home/pi/Desktop/cartdump.txt
Traceback (most recent call last):
  File "/usr/local/bin/stratatools", line 9, in <module>
    load_entry_point('stratatools==3.0', 'console_scripts', 'stratatools')()
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 217, in main
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 31, in run
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 131, in command_eeprom_decode
  File "build/bdist.linux-armv7l/egg/stratatools/manager.py", line 54, in decode
  File "build/bdist.linux-armv7l/egg/stratatools/manager.py", line 112, in unpack
Exception: invalid content checksum: should have 0xdd47 but have 0xd7d2
```

williamhmic, 2018-09-21:
@bvanheu @HaveBlueXST Hello? Any ideas? Please?

HaveBlueXST, 2018-09-30:
Sorry, I re-tasked my Raspi to another project, but hopefully I can get another and try things for myself with the latest tools and OS image.