bvanheu/stratatools #67: replacement parts- I know this is not the right place for this

calvindomenico, 2018-08-03:
Does anyone have a source for them ? or if someone know the exact spec I can just source them 
after a terrible jam that left my head full fo ABS, I need to replace my cartridge heaters.

I have found replacement thermocouples that are only about $35 each if anyone is interested
Mineral Insulated Metal Sheath Thermocouple Probes Bare Leads Mineral Insulated MI thermocouple material Cable MIC replacement BLMI TC probe
Part number 
BLMI-INC-K-116U-12

I have used these before so I know they fit and work 

Thank You
Calvin

tjschweizer, 2018-08-05:
What kind of printer?

calvindomenico, 2018-08-05:
Fortus 250 

Thanks for getting back to me

C

AaronK43, 2018-09-27:
Hi Calvin,

Were you able ti find the cartridge heaters?  I have access to many Stratasys parts and can help you with any sourcing you might need.  Please feel free to send me your email address and I can provide further info.

Thanks