bvanheu/stratatools #73: Stratasys Dimension 1200es Firmware Update Safe?

quark78, 2018-12-17:
I currently have Version 7.0 Build 1866 firmware running on my Dimension 1200es (firmware circa 2007). I see that the latest available firmware is Version 10.1 Build 3810 (firmware circa the past ~2 years or so I think). I currently use this eeprom programmer to reset/reload both the model and support cartridges (Filling them to 99,999 in^3). Does the latest 10.1 firmware still accept rewritten eeproms on both the cartridges (and to 99,999 in^3)? Can anyone confirm this?

azumi-san, 2019-01-14:
For rewritten eeproms - yes, it works on 10,1 firmware.
For 99,999 in^3 - don't know.

CTSchorsch, 2019-04-01:
On my machine "Dimension Elite" with 10.1 Build 3810 the rewritten cartridges are not working. "Material Error". I use the daemon mode to refill the cartridge. 
@azumi-san  how do you rewrite your eeproms ?

azumi-san, 2019-04-01:
> On my machine "Dimension Elite" with 10.1 Build 3810 the rewritten cartridges are not working. "Material Error". I use the daemon mode to refill the cartridge.
> @azumi-san how do you rewrite your eeproms ?

I'm using C# version of flashing software for my 1200es.
Select Prodigy machine.
Read it at first, then just change material quantity and serial number. 
Try to use quantity less than 60in^3.

CTSchorsch, 2019-04-02:
Thanks, the arduino/windows workflow works here also.

bvanheu, 2019-05-06:
@CTSchorsch do you know if the issue with the daemon mode was related to the qty of material on the cartridge?

CTSchorsch, 2019-05-06:
@bvanheu  I dont know. I switched to the windows/arduino version and did not investigate the problems with the linux daemon

davey242, 2024-11-23:
Where can I download the latest firmware? Or v10.1 mentioned here?
I just got one of a bst 1200 for free and figured I would update from 5.4 if I can