bvanheu/stratatools #74: Does this software work with the Fortus 250mc?

joesgarage, 2019-05-05:
Hi I very new here.   Will this command line work for the fortus 250mc?    I keep getting usage errors.

bvanheu, 2019-05-05:
Can you paste your command line?

What is the usage error printed on the console?

joesgarage, 2019-05-05:
Hello,  and thank you for responding.  I'm not sure if Im doing this right - at all...
 Im able to read and see the eeprom show up in w1_bus_master.   So if I execute the following command
stratatools eeprom_decode -t 250mc -e 5edce801000023 cartridge_dump.bin
I get: 
usage: stratatools eeprom_decode [-h] -t
                                 {uprint,ktype,fox,prodigy,quantum,fox2} -e
                                 EEPROM_UID [-D]
                                 [input_file] [output_file]
stratatools eeprom_decode: error: argument -t/--machine-type: invalid choice: '250mc' (choose from 'uprint', 'ktype', 'fox', 'prodigy', 'quantum', 'fox2')

neither fortus or 250mc show up as an option.

Im just trying to understand if the 250mc is supported and what the steps are to re write this eeprom with a new material value (and serial # ? if required?).
Please.

bvanheu, 2019-05-05:
The error is:

```
error: argument -t/--machine-type: invalid choice: '250mc' (choose from 'uprint', 'ktype', 'fox', 'prodigy', 'quantum', 'fox2')
```

The machine type parameter `-t` takes only one of the following value:

- uprint
- ktype
- fox
- prodigy
- quantum
- fox2

HaveBlueXST, 2019-05-05:
> neither fortus or 250mc show up as an option.
> 
> Im just trying to understand if the 250mc is supported and what the steps are to re write this eeprom with a new material value (and serial # ? if required?).

I believe the Fortus 2xx series are all P-class machines, so you should be able to use the 'prodigy' type.

joesgarage, 2019-05-05:
thank you for your prompt responses.   I'm still not getting anywhere. 
I went thru the readme but cant get anything to work. 
What are the steps to read the eeprom, change the material % value, and write it back ?

HaveBlueXST, 2019-05-05:
> thank you for your prompt responses. I'm still not getting anywhere.
> I went thru the readme but cant get anything to work.
> What are the steps to read the eeprom, change the material % value, and write it back ?

You can see examples in my blog post (jump about 2/3 of the way down): https://haveblue.org/?p=1988

joesgarage, 2019-05-05:
Ok, that was the original blog I was reading that lead me to this github page. But I am trying to change the values and create a new .bin file for which I will upload to the eeprom. but when I execute
stratasys-cli.py
it says command not found . (this is per your blog page).

bvanheu, 2019-05-06:
I've updated the CLI interface to make it easier to use, but the idea is the same as the blogpost by Have Blue.

I think you were on the right way, this should work:
```
stratatools eeprom_decode -t prodigy -e 5edce801000023 cartridge_dump.bin
```

joesgarage, 2019-05-06:
Hi !  thank you again for your time.  
Yes, I did figure that out after I sent it. 
So I can read and create a bin file, and then display the contents using:
stratatools eeprom_decode -t prodigy -e  14000001e8dc5e23 eeprom.bin
but what switches do I need to reset the serial and amount ?
eeprom_encode ?  or _create ?  and with what commandline switches ?

bvanheu, 2019-05-06:
you'll typically use it this way:
```
$ stratatools eeprom_create --serial-number 30035862.0 --initial-material 56.3 --current-material 56.3 --version 1 --signature STRATASYS > eeprom_template.txt
$ stratatools eeprom_decode -t prodigy -e 5edce801000023 eeprom_template.txt > eeprom.bin
```

Or you can do both on one line:
```
$ stratatools eeprom_create --serial-number 30035862.0 --initial-material 56.3 --current-material 56.3 --version 1 --signature STRATASYS | stratatools eeprom_decode -t prodigy -e 5edce801000023 > eeprom.bin
```

The command `eeprom_create` will create a structure to represent the cartridge (e.g. material, quantity, etc). You then need to "encode" it for a specific printer using `eeprom_encode`.

joesgarage, 2019-05-06:
almost there, I think, but:

stratatools eeprom_create --serial-number 30035862.0 --initial-material 56.3 --current-material 56.3 --version 1 --signature STRATASYS > eeprom_template.txt

Traceback (most recent call last):
  File "/usr/local/bin/stratatools", line 11, in <module>
    load_entry_point('stratatools==3.0', 'console_scripts', 'stratatools')()
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 217, in main
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 31, in run
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 141, in command_eeprom_create
  File "/usr/local/lib/python2.7/dist-packages/protobuf-3.7.1-py2.7.egg/google/protobuf/internal/python_message.py", line 716, in field_setter
    new_value = type_checker.CheckValue(new_value)
  File "/usr/local/lib/python2.7/dist-packages/protobuf-3.7.1-py2.7.egg/google/protobuf/internal/type_checkers.py", line 177, in CheckValue
    raise TypeError(message)
TypeError: None has type <type 'NoneType'>, but expected one of: (<type 'str'>, <type 'unicode'>)

bvanheu, 2019-05-06:
Received this by email
```
almost there, I think, but:

stratatools eeprom_create --serial-number 30035862.0 --initial-material 56.3 --current-material 56.3 --version 1 --signature STRATASYS > eeprom_template.txt

Traceback (most recent call last):
File "/usr/local/bin/stratatools", line 11, in 
load_entry_point('stratatools==3.0', 'console_scripts', 'stratatools')()
File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 217, in main
File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 31, in run
File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 141, in command_eeprom_create
File "/usr/local/lib/python2.7/dist-packages/protobuf-3.7.1-py2.7.egg/google/protobuf/internal/python_message.py", line 716, in field_setter
new_value = type_checker.CheckValue(new_value)
File "/usr/local/lib/python2.7/dist-packages/protobuf-3.7.1-py2.7.egg/google/protobuf/internal/type_checkers.py", line 177, in CheckValue
raise TypeError(message)
TypeError: None has type <type 'NoneType'>, but expected one of: (<type 'str'>, <type 'unicode'>)
```

Make sure you pass every parameter to `eeprom_create`, it might be the issue here.

I should probably set some default value :)

```
stratatools eeprom_create --material-name ABS --manufacturing-lot 123.0 --manufacturing-date "2019-05-06 12:00:00" --use-date "2019-05-06 12:00:01" --initial-material 56.3 --current-material 56.2 --key-fragment 4141414141414141 --serial-number 30035862.0 --version 1 --signature STRATASYS
```

joesgarage, 2019-05-07:
Ok.  I got it to work !   @bvanheu @HaveBlueXST   Thank you for your help and your patience with me !!
I will post all my command lines if its ok, in case someone lands here the way I did, looking for info ?

1 last question, please.

In the example you gave, the command line switches differ in format from that of the help in stratatools, where you use  -- 
In your example you use:  --initial-material 56.3
why don't you use:     -n 56.3
as shown in the help menu:  stratatools eeprom_create -h   ??

bvanheu, 2019-05-08:
`-n` is called  a "short switch" and `--initial-material` is called a "long switch". Both are equivalent, sometimes it's easier to remember the long switch, and usually more clear what's the purpose of the switch compared to the short one!

closed 2019-06-30 by bvanheu