bvanheu/stratatools #75: No eeprom File

Hans-kloss2, 2019-06-12:
I have a normal eeprom, but there is no eeprom file.
And with the automation of the raspberry pi, he can't find the eeprom file (stratatools_rpi_daemon prodigy)
are there someone who can help me?

Hans-kloss2, 2019-06-12:
the eeprom are making some repositories without the eeprom file
[image]
[image]
[image]

Hans-kloss2, 2019-06-12:
please help

Hans-kloss2, 2019-06-12:
for more informations just ask

Hans-kloss2, 2019-06-13:
i have an eeprom from stratasys (sst 768)

bvanheu, 2019-06-30:
Likely a problem with the physical connection with the EEPROM. Too long wire maybe?

Hans-kloss2, 2019-07-07:
thanks, i will try it

Hans-kloss2, 2019-07-07:
no it did not work, i tried with a shorter wire. i have still the same problem. there is no eeprom file to manipulate :(

Hans-kloss2, 2019-07-07:
[image]
[image]
Do you know, where the issue is. because i tried it with shorter wire didn't work. and on JPG 0700 what does the programm means with machine type?

Hans-kloss2, 2019-07-07:
i have a stratasys (sst 768) eeprom, do you think they have changed the software on the eeprom?

Hans-kloss2, 2019-07-07:
[image]
the blue one. in the repository is no eeprom file to manipulate. i think they have changed the software from the eeprom. do you think i just can at my own make an eeprom file, and copy it into it?

Hans-kloss2, 2019-07-07:
sorry the purple circle

bvanheu, 2019-07-07:
Can you give me the output of this command:
`ls /sys/bus/w1/devices/`

while the eeprom is connected

Hans-kloss2, 2019-07-08:
[image]
the output is (w1_bus_master1)
i tried it a few more times.

Hans-kloss2, 2019-07-08:
[image]
now there are more folders?

Hans-kloss2, 2019-07-08:
[image]
after some refreshings, there are more folders then just the w1_bus_master1 folder, after about 1 minute the folders changed? (Name?) please help :(
[image]

ltmetalman, 2019-08-18:
I am getting this too. I tried putting the positive at the 3.3 volt source and I got a family code of 68 which didn't make sense. Haven't been able to reproduce that though.

Hans-kloss2, 2019-08-18:
i'm making it on a raspberry pi 2 b with noobs 1.7

Hans-kloss2, 2019-08-18:
but i think that my eeprom is a newer one, and the software does not recognize it.

Hans-kloss2, 2019-08-18:
i think our eeproms aren't supported as you said