bvanheu/stratatools #76: UprintSE checksum

camosquirrel, 2019-08-14:
Others have mentioned checksum errors when decoding UprintSE eeproms, as in issues [52](https://github.com/bvanheu/stratatools/issues/52) and [55](https://github.com/bvanheu/stratatools/issues/55). It doesn't work for me either. Can anyone else decode the eeprom (attached, [eeprom.log](https://github.com/bvanheu/stratatools/files/3499598/eeprom.log)) with carrier id `b3 37 bd 0b d0 14 10 c8` to confirm that the problem is specific to me?

Traceback:

user@host:~/Desktop/stratasys/stratasys-master$ python stratasys-cartridge.py info -t uprintse -e c81014d00bbd37b3 -i ../test.bin
Traceback (most recent call last):
  File "stratasys-cartridge.py", line 204, in <module>
    app.run()
  File "stratasys-cartridge.py", line 47, in run
    args.func(args)
  File "stratasys-cartridge.py", line 134, in command_info
    cartridge = m.decode(machine_number, args.eeprom_uid, cartridge_crypted)
  File "/home/user/Desktop/stratasys/stratasys-master/stratasys/manager.py", line 74, in decode
    cartridge = self.unpack(cartridge_packed)
  File "/home/user/Desktop/stratasys/stratasys-master/stratasys/manager.py", line 130, in unpack
    raise Exception("invalid content checksum: should have " + hex(struct.unpack("<H", str(cartridge_packed[0x40:0x42]))[0]) + " but have " + hex(self.checksum.checksum(cartridge_packed[0x00:0x40])))
Exception: invalid content checksum: should have 0x9806 but have 0xbc21

I'm reading the chip through the DIAG port, I confirmed with a microscope it is a DS2432. Here's the data by serial port:

```
er 0 0 0 128
Model carrier ID (8 8-bit values)
000000: b3 37 bd 0b d0 14 10 c8                           .7......

Model carrier in bay 0: 128 bytes at address 0 (128 8-bit values)
000000: 21 3c ef f4 cf 81 24 cb a3 13 f7 5b 4d 72 d9 c9   !<....$....[Mr..
000016: fb e5 b5 26 3e 96 f4 7d d8 6a ee 3c 2c 4c 2e 32   ...&>..}.j.<,L.2
000032: dd f8 6a 3f 49 61 cd db a4 3a c3 26 c9 93 e9 d1   ..j?Ia...:.&....
000048: a4 3a c3 26 c9 93 e9 d1 2a f9 e2 04 b2 a9 b2 cc   .:.&....*.......
000064: 06 98 00 00 00 00 83 73 55 aa 55 c2 e3 52 4b be   .......sU.U..RK.
000080: 9a 08 00 00 00 00 00 00 49 d8 20 57 e9 2b 6c 98   ........I. W.+l.
000096: 86 05 8c 4b 01 00 00 00 53 54 52 41 54 41 53 59   ...K....STRATASY
000112: 53 ba 2c 9c f4 62 35 27 1f 42 88 e2 cb 44 48 e6   S.,..b5'.B...DH.
```

It may be relevant that the in the referenced issues, the people experiencing checksum fails are also running linux.

bvanheu, 2019-08-14:
I think the eeprom id is in the wrong byte order. Try this:
```
python stratasys-cartridge.py info -t uprintse -e b337bd0bd01410c8 -i ../test.bin
```

camosquirrel, 2019-08-14:
The checksum still appears to be invalid -

```
python stratasys-cartridge.py info -t uprintse -e b337bd0bd01410c8 -i ../test.bin
Traceback (most recent call last):
  File "stratasys-cartridge.py", line 204, in <module>
    app.run()
  File "stratasys-cartridge.py", line 47, in run
    args.func(args)
  File "stratasys-cartridge.py", line 134, in command_info
    cartridge = m.decode(machine_number, args.eeprom_uid, cartridge_crypted)
  File "/home/user/Desktop/stratasys/stratasys-master/stratasys/manager.py", line 74, in decode
    cartridge = self.unpack(cartridge_packed)
  File "/home/user/Desktop/stratasys/stratasys-master/stratasys/manager.py", line 130, in unpack
    raise Exception("invalid content checksum: should have " + hex(struct.unpack("<H", str(cartridge_packed[0x40:0x42]))[0]) + " but have " + hex(self.checksum.checksum(cartridge_packed[0x00:0x40])))
Exception: invalid content checksum: should have 0x9806 but have 0xb991
```

Can anyone successfully decrypt the posted eeprom binary? This would establish that the error is particular to me rather than a software bug.

TheGuardDawg, 2020-03-23:
Sorry to interrupt 😅 I’ve accidentally locked my uprint canister chips, wondering if anyone might have a couple near empties they’d be willing to send over. Thank you!