bvanheu/stratatools #78: Dimension printer automation?

GilDev, 2019-12-05:
Hi,

How should I proceed with a Stratasys Dimension printer? It seems to decode correctly when I set it to “prodigy” printer type, but won’t work with other types.
The problem is with the automation script, here is what I get when I connect a cartridge’s EEPROM:

```
New device detected <23d985710100006f>.
Error! verify machine type?
Details:
Traceback (most recent call last):
  File "build/bdist.linux-armv6l/egg/stratatools/helper/rpi_daemon.py", line 37, in on_new_cartridge
    c = cartridge_manager.decode(machine_number, eeprom_uid, read_bytes(eeprom_path))
  File "build/bdist.linux-armv6l/egg/stratatools/manager.py", line 54, in decode
    cartridge = self.unpack(cartridge_packed)
  File "build/bdist.linux-armv6l/egg/stratatools/manager.py", line 112, in unpack
    raise Exception("invalid content checksum: should have " + hex(struct.unpack("<H", str(cartridge_packed[0x40:0x42]))[0]) + " but have " + hex(self.checksum.checksum(cartridge_packed[0x00:0x40])))
Exception: invalid content checksum: should have 0xf900 but have 0x9e0c
```

GilDev, 2019-12-05:
I’m trying a few things to try to make it work and I’ve rewritten the `on_new_cartridge()` function of the `helper/rpi_daemon.py` file.
When calling `cp ~/eeprom_new.bin /sys/bus/w1/devices/w1_bus_master1/23-xxxxxxxxxxxx/eeprom`, I randomly get this error:
`cp: error writing '/sys/bus/w1/devices/w1_bus_master1/23-xxxxxxxxxxxx/eeprom': Input/output error`

Why is this happening? It’s the only thing that’s blocking me right now. If it can be fixed I’ll post my working script.

mateosar, 2020-02-04:
> Hi,
> 
> How should I proceed with a Stratasys Dimension printer? It seems to decode correctly when I set it to “prodigy” printer type, but won’t work with other types.
> The problem is with the automation script, here is what I get when I connect a cartridge’s EEPROM:
> 
> ```
> New device detected <23d985710100006f>.
> Error! verify machine type?
> Details:
> Traceback (most recent call last):
>   File "build/bdist.linux-armv6l/egg/stratatools/helper/rpi_daemon.py", line 37, in on_new_cartridge
>     c = cartridge_manager.decode(machine_number, eeprom_uid, read_bytes(eeprom_path))
>   File "build/bdist.linux-armv6l/egg/stratatools/manager.py", line 54, in decode
>     cartridge = self.unpack(cartridge_packed)
>   File "build/bdist.linux-armv6l/egg/stratatools/manager.py", line 112, in unpack
>     raise Exception("invalid content checksum: should have " + hex(struct.unpack("<H", str(cartridge_packed[0x40:0x42]))[0]) + " but have " + hex(self.checksum.checksum(cartridge_packed[0x00:0x40])))
> Exception: invalid content checksum: should have 0xf900 but have 0x9e0c
> ```
> 
> Do you have an idea on what I could do to make it work? Thanks!

Where you able to solve the Check sum error? I am having the same issue

GilDev, 2020-02-04:
No I haven't worked on it again but am still very interested about resolving the issue!

SOCIT, 2020-02-07:
> How should I proceed with a Stratasys Dimension printer? It seems to decode correctly when I set it to “prodigy” printer type, but won’t work with other types.

Dimension uses the "prodigy" type.   What specifically are you trying when you get the checksum error?