bvanheu/stratatools #80: Reading F-Series chips? (F120, F270, F270, F370)

Tomblarom, 2020-04-20:
My university got an Stratasys F370. Since I created [this resetter](https://www.thingiverse.com/thing:2636435) for the Dimension Elite for my former school, I was curious, if these chips can also be manipulated. Couldn't find anything about them.

I've two EEPROMS. One totally empty 0% and one on 2%:
[image]

On the EEPROM is the following written:

    40+      40+
    1101     1101
    917A3    914A3

Some people asked for this as well:
- 
- https://haveblue.org/?p=1988&cpage=2#comment-1000

I found this EEPROM, which is pretty similar to which Stratasys uses:
https://de.rs-online.com/web/p/eeproms/7327539/

luizrenault, 2021-09-30:
Any luck?
I have the same and wasn't able to access the chip using w1-gpio driver on a RPI.

But with some modifications, I can now access it. It says family 0C. Any ideas?

galicminer, 2022-07-02:
@luizrenault do how did you get to read the Eeprom, my raspberry refuses to recognize it.
did you manage to get a dump of yours?

pkoprov, 2022-12-13:
Same stuff. What modifications did you make?

pkoprov, 2022-12-13:
> Any luck? I have the same and wasn't able to access the chip using w1-gpio driver on a RPI.
> 
> But with some modifications, I can now access it. It says family 0C. Any ideas?

Same stuff. What modifications did you make?

shisva, 2023-08-25:
Hi any news on this topic ?

Tomblarom, 2024-02-09:
Here somebody was looking for the same chip: https://www.reddit.com/r/AskElectronics/comments/oewa1l/help_identifying_ic_believe_to_be_some_kind_of_3/

Zannys, 2024-02-10:
> Here somebody was looking for the same chip: https://www.reddit.com/r/AskElectronics/comments/oewa1l/help_identifying_ic_believe_to_be_some_kind_of_3/
> 
> There are chips which are very similar and have been suspectivly identified as **DS2430 / DS2431**: `40+ 1101 917A3` Stratasys F370 `2D 1002 926B1` Discus Dental ZOOM 2600
> 
> > It says family 0C
> 
> @luizrenault if you are correct, then it might be the same chip that was used in **iButton 1996** devices: https://owfs.sourceforge.net/family.html Here is a disassembly of these devices: https://murdoch.is/projects/ibutton/
> 
> [Analog_DS1996.pdf](https://github.com/bvanheu/stratatools/files/14220108/Analog_DS1996.pdf) [DALLAS_DS1996.pdf](https://github.com/bvanheu/stratatools/files/14220110/DALLAS_DS1996.pdf)

Any news? i tried and don't even can't even make the raspberry identified IC

Tomblarom, 2024-02-14:
> But with some modifications, I can now access it. It says family 0C. Any ideas?
@luizrenault can you please elaborate how you did it esthablish a connection? How does the wiring look like?

skylatian, 2024-02-21:
@luizrenault, any word on what those "changes" were? My university would *love* to be able to use our own filament in these machines.

tkrsntr, 2024-04-24:
@luizrenault, I'm also curious about how you read the chip using RPI, what modifications are made?

Kiano950, 2025-02-27:
Are there any updates?

camtheham, 2025-03-27:
> Any luck? I have the same and wasn't able to access the chip using w1-gpio driver on a RPI.
> 
> But with some modifications, I can now access it. It says family 0C. Any ideas?

When I set up the circuit with an Arduino and do a .search() with the OneWire library I am getting a family code of 3A. 

According to this site, that family code is a DS2413. This is a 1-Wire Dual Channel Addressable Switch. I don't quite see how an addressable switch would be used as a spool cartridge tracker though. 

https://owfs.org/index_php_page_family-code-lookup.html

ashaninn, 2025-05-03:
If anyone is still interested, we have a solution. Not free, but it will save you alot of money. (you can use your eeproms forever)
Also there is a solution that will upgrade any f123 to f370CR. Please send me email with your questions to ssysusr@gmail.com

luizrenault, 2025-06-27:
It is cheapest to get a K2 Plus than dedicating hours working on this matter.

dan1282, 2025-07-25:
As far as I'm aware, there are two solutions currently available.  Argyle materials partnered with someone that came up with a way of resetting chips.  They initially released a solution last year but it required you to have the filament spool outside of the machine and feed it into the back.  It looks like they've recently improved that solution so you can now put a 184ci spool directly in the printer bay and then there's a usb cable the comes out and connects to the front of the F370.  Not sure if this is for power or communication or both but with this solution you are buying filament directly from them at a 21% discount over stratasys filament.  They're running a promotion until the end of July that brings the cost down to about a 50% savings.  Still expensive but much cheaper than Stratasys.  We may end up getting a spool to see how it goes.  The other company (who i believe is the person that posted above) i a full sw unlock that allows you to run your own filament.  https://www.printoverride.com/  was the website but it's not active yet but they are selling on ebay https://www.ebay.com/itm/197339596259?_skw=stratasys+f123+eeprom  $2499 for the unlock.  Steep price but can pay for itself in basically 10 spools.

wowgrant, 2025-07-25:
Yes, I am familiar with the PrintOverride team, so I can give some context.

Their solution is a full software unlock — no add‑on boards, no cables, no external power. Everything runs natively inside the printer, and it’s 100 % reversible: if needed, you can roll back to the factory firmware in a few minutes.

They also added something pretty unique:
Custom initial volume assignment for any chip (e.g., 200 ci, 500 ci — whatever you want). This value is set during the upgrade process. Each time you reinsert the spool, it resets to that value. The printer still tracks real depletion during printing, and because the F‑series has proper filament sensors, it handles bay auto‑switching like stock.
Importantly, the printer can read any old chip, even fully depleted ones — the material type remains untouched, so the printer still recognizes the original material type, but always sees the chip as new and full, regardless of its actual previous state. Since the F‑series has proper filament sensors, it also handles bay auto‑switching like stock.

They’ve also created a material bay model for standard 2 kg spools (around 123 ci), making it much easier to use non‑OEM filament.

As for cost — it’s really not high when you look at ROI. Just a few spools are enough to pay off the upgrade, and from there you fully control your machine instead of being locked to Stratasys or resellers like Argyle. And with Argyle, you’re still tied to their material supply (basically just another vendor), with pricing only slightly better than OEM.

They’ve been running this on their own fleet (F370CRs + Fortus 450s), printing all kinds of ASA, PC‑ABS, and TPU blends at a fraction of OEM cost. They’re also actively experimenting with more advanced materials like PC, PPSF, and PSU — though those require additional hardware considerations.

For anyone running real production, the difference compared to vendor‑locked setups is night and day. If anyone wants technical details, feel free.

wowgrant, 2025-07-26:
Thanks for clarifying — that actually makes a lot of sense.

The fear of “unknowns” is common, but in practice the unlock doesn’t change any of the core mechanics of the machine. The print process itself is untouched — same motion system, same extrusion, same chamber control. What you gain is simply the ability to load and run whatever material you want without the printer blocking you.

On mixing OEM + third‑party: no problem at all. You can keep using OEM where it makes sense (like QSR/SR‑30 support), and run third‑party for model material in the other bays. The printer handles them side by side without issue. For soluble supports — QSR (actually SR‑35) and SR‑30 are interchangeable, and for many geometries we’ve also successfully used HIPS as a low‑cost support. In fact, in most cases HIPS works much better than QSR: it acts like a breakaway support. Since the majority of printed parts aren’t complex biomimetic structures with fully enclosed cavities, simply breaking off the support is often far quicker and cleaner than soaking the entire model for hours, rinsing, and drying.

As for materials: one big advantage is you can switch to low‑cost but branded filaments like BambuLab or industrial‑grade ASA/PC/PC-ABS blends from large suppliers. Stratasys itself doesn’t manufacture filament — they source blends from global giants like SABIC. We’ve done extensive testing and selected materials that actually perform as well as or better than OEM on F‑series and Fortus machines. If clogging or print failures are a concern — that’s why we recommend sticking to reputable brands. Even budget lines from companies like BambuLab are clean, consistent, and free of debris or fillers that could block a nozzle.

Additionally, for our own production we’ve developed heavy‑duty “lifetime” print heads to replace the OEM ones, which we found poorly designed and short‑lived. Our heads use tungsten‑carbide rollers, ceramic and bronze gears, and precision bearings, giving them a lifespan of 15 000–20 000 hours versus 750–1 000 hours for stock units. They deliver more consistent extrusion, handle abrasive filaments without degradation, and eliminate the recurring downtime and cost of swapping heads.

We’ve been running this setup on our own F‑series and Fortus machines for years under production loads — deadlines, long runs, zero tolerance for downtime. It’s been rock solid, and the nice thing is the unlock is fully reversible: if you ever need to roll back to OEM‑only mode, it’s a 5‑minute process.

dan1282, 2025-07-26:
Wow super helpful. Sounds like you guys have done a lot in developing this solution. I’d love to see a list of materials that you’d recommend running on the f series. We exclusively run asa just bc it’s easy to print, has UV stability (something important in our field) and is durable. I’d also love to know when you think you’ll be offering the print heads and approx cost. Longer lifetime sounds like a dream. We frequently overrun the stratasys heads but also see degradation in print quality/stringing over time. We’re in a heavy development cycle right now with the printer running almost constantly so are blowing through filament and it’d be great to cost reduce where we can. I’m seriously considering skipping argyle completely and just biting the bullet with your solution. We just did 2 head swaps as well so might be a great time.

wowgrant, 2025-07-26:
Glad it helps — that’s exactly why we’ve invested so much into making this a full solution, not just a bypass.

On materials: we do have a list of options that work reliably on the F‑series , feel free to PM.

On print heads: we’ve designed and use our own mechanical assemblies in daily production, replacing the weak OEM drive train with four tungsten‑carbide rollers for maximum grip, ceramic/bronze gearing, and precision bearings. These heads are far more durable and maintain consistent extrusion even under heavy loads. They’re already in operation on our machines and can be built to order for others.

Additionally, we can adjust the printer’s internal head lifetime counters as part of the upgrade — so the system recognizes the new head without throwing errors or service warnings.

dan1282, 2025-08-25:
So we pulled the trigger on this upgrade and after a few days back and forth with the print override team, we have a fully unlocked machine.  My understanding is that the original file didn't work bc essentially we purchased one of the first F370's as soon as they came out and it had an older bootloader than what this upgrade was developed on.  Peter was not only extremely responsive but very through and clearly knew this machine in and out.  After running a few tests, he was able to isolate the problem and provide us with a fix that works just as good (if not better than their commercial solution).  We've already starting printing in bambu ASA and only 1 print so far but it works seamlessly.  We will probably purchase the upgraded print heads once our current ones run through their ~1500hrs.  Throughout the time troubleshooting the issue we had, it became very apparent that this is a very well thought out and well developed unlock.  They have been working on it for years and have had several iterations before they came to the commercial release that's being provided today.  They have also run 10+ tons of material through their own printers using this method.  While i may have been a little hesitant at first, I have to say my confidence with this has grown exponentially.  I fully expect to see the cost paid for within 2-3 months if not sooner.

arnoliudaxia, 2025-11-07:
> Any luck? I have the same and wasn't able to access the chip using w1-gpio driver on a RPI.
> 
> But with some modifications, I can now access it. It says family 0C. Any ideas?

I use rasp 4b , also failed to read, only show family code 0x00, ghost eeprom

fortususer, 2026-09-14:
We ended up solving this from the printer side rather than by reading or rewriting the F-Series material EEPROMs.

NEXT 3DP now has a printer-side material unlock for the F120, F170, F270, F370 and F370CR. It allows compatible/refill material workflows without an external EEPROM writer or powered material-bay dongle.

Activation is applied by USB in about a minute and is tied to the printer serial number.

Posting this here because it may be useful to anyone who found this issue while looking for a practical way around the F123 material restrictions.

https://www.next3dp.com/order/p/f123-material-unlock

Disclosure: this is our commercial solution.

wowgrant, 2026-09-14:
Relying on a permanently attached USB drive to execute runtime scripts on a live, booting controller is an extremely fragile, basement-tier workaround rather than a genuine printer-side solution.

This is essentially the software equivalent of the old hardware "dongle rescue devices" that sat as an awkward shim between the material chip and the bay contacts. It trades one physical point of failure for another, introducing unacceptable risks for any commercial or corporate environment:

High Risk of Bricking the Controller: Executing runtime patches via external drive mounts while core Stratasys daemons and hardware watchdogs are initializing is a reliable way to corrupt local configurations, break native services, or leave the embedded controller completely unbootable. Recovering a bricked F123 single-board computer is a costly maintenance nightmare.

Physical Hardware & Interface Fragility: Keeping a consumer USB flash drive permanently plugged into an industrial machine exposes the controller to vibration, port fatigue, contact wear, and silent read errors. If the drive drops out mid-cycle or fails to mount cleanly during boot, the printer immediately faults into a factory lockout.

Zero Deterministic Stability: Because an ephemeral drive-dependent hook reverts with every reboot or power drop, there is zero persistent system integrity. One transient brownout or controller restart mid-shift can instantly derail an active production run.

If an individual hobbyist picked up an old machine for pennies to tinker in a garage with scrap prints, hanging an external flash drive off the controller might seem like an amusing shortcut. But for corporations, labs, and production bureaus running critical jobs, relying on an unverified USB dongle to inject code into a live system is an unnecessary gamble. Reliable operation requires clean, stable, native software architecture, but not consumer flash drives left hanging off industrial hardware.

fortususer, 2026-09-14:
Just to clarify — there is no USB drive permanently attached to the printer.

USB is only used to install the upgrade and is removed afterward. The solution then runs entirely printer-side with no external drive, dongle, or mounted USB dependency during normal operation.

So the vibration, USB-port fatigue, mount failure, and “drive drops out mid-print” concerns described above simply don’t apply to our implementation.

wowgrant, 2026-09-14:
That phrasing directly contradicts the technical documentation on your own site, which states:

_"The unlock does not permanently modify the printer's onboard firmware or software... If the printer is restarted without the drive, it simply operates as a standard factory F123 with normal material restrictions."_

Whether the drive remains physically seated during operation or an operator is required to walk up and plug external media back into the machine on every reboot or power drop, the underlying reality is the same: it is an ephemeral, volatile script injection triggered from external media rather than a persistent, integrated platform solution.

From an embedded engineering and shop-floor standpoint, this introduces critical liabilities:

In-Memory Race Conditions: Running volatile startup scripts against live daemons without deterministic system integration invites race conditions. If native background processes, serial handlers, or controller integrity checks fire out of sequence before the script cleanly executes, the system risks entering an unrecoverable fault state or corrupting core configurations.

Operational Unreliability: Relying on technicians to manually cycle external USB drives to keep machines functional after scheduled maintenance, power flickers, or thermal reboots is unacceptable in a production environment.

Warranty & Telemetry Exposure: High-tier production systems like the F370 or F770 constantly log abnormal filesystem mounts, unexpected process hooks, and mid-run fault interruptions. Clumsy runtime injections leave blatant forensic footprints in system diagnostics, instantly voiding active service contracts and guaranteeing support refusal from Stratasys.

Unacceptable Risk Profile: No qualified manufacturing engineer or lab technician in their right mind would risk a $50,000–$100,000 production asset like an F370 or F770 on unverified script execution. The cost of bricking an embedded single-board computer or halting a critical 60-hour build vastly outweighs any perceived convenience.

A true machine-side solution must be deterministic, cleanly integrated, and persistent across reboots without requiring recurring physical media hand-holding or exposing high-value industrial hardware to runtime instability.

"Vibecoding" and stitching together transient shell scripts might be fun for side projects, but it has no place in the control loop of mission-critical industrial hardware.

fortususer, 2026-09-15:
You’re still assigning failure modes to the architecture that don’t follow from how it operates.

The UPG is executed through the normal Stratasys software-update mechanism. It is not a startup script racing the controller’s initialization sequence.

If the modification fails or the controller reboots, the result is the opposite of a persistent brick — the modification is gone and the machine boots stock again. A controller reboot during a 60-hour build is already a production interruption regardless of whether this unlock exists.

For this development we processed 100’s of UPGs and experienced zero controller issues. So claims of likely filesystem corruption, controller bricking, or “unrecoverable” race conditions simply don’t reflect what we have actually observed.

The real tradeoff is straightforward: after a controller reboot, the operator reapplies the unlock if they want it. The entire process takes less than a minute…

wowgrant, 2026-09-15:
Thank you for confirming the exact operational model. Admitting that the UPG bypasses persistence and immediately reverts upon any controller reboot makes the underlying technical compromise transparent:

Inability to Recreate the Chain of Trust: Because your method cannot validate or reconstruct the platform's hardware root-of-trust and secure verification chain, it forces the operating system to run in a volatile, highly unstable state. The moment an unexpected deviation, signature mismatch, or background system check occurs, the controller faults out, carrying an unacceptable risk of bricking the hardware.

Continuous eMMC / NAND Memory Wear: Triggering recurring UPG installations (which extract archives, write temporary data, and cycle blocks on the embedded storage) every single time a machine is rebooted subjects the onboard flash memory to continuous write wear. Embedded industrial boards have finite write endurance; abusing the update mechanism as a recurring daily launcher is terrible engineering practice.

Complete Infeasibility in Production: No sane manufacturing engineer or lab technician in a professional environment is going to walk over with a USB flash drive to re-flash the controller through the touch screen after every routine restart, maintenance window, or power drop. If a tool cannot maintain its operational state deterministically without manual human babysitting on every boot, it simply cannot be deployed in serious production.

Flashing 100 Times ≠ 100 Real Machines: Claiming you “processed 100’s of UPGs” during development just means you ran repeated test cycles on a couple of bench units. Manually installing test packages on a workbench has nothing to do with deploying, supporting, and maintaining a mission-critical fleet of 100+ production-grade machines running multi-day industrial builds under real pressure.

The F770 Reality: Stating on your site that the F770 isn't supported because of "different material handling" completely misses the mark. The restriction isn't about the materials - it's because the underlying architecture and base operating environment are different. When a solution relies on surface-level script workarounds instead of deep architectural comprehension, scaling to that tier is simply out of reach.

This is precisely why our platform took a completely different path long ago. We don't rely on volatile mounts, USB dependencies, or transient user-space restarts that degrade flash memory.

Instead, we built an integrated, production-grade ecosystem:

**Full Chain-of-Trust Integrity:** A complete, persistent operating system environment that functions exactly as the OEM originally architected, but in its unrestricted, fully expanded state.

**Native Ecosystem & Dedicated Software:** Our proprietary suite interfaces directly with the printer's core operational logic, providing machine-side parameter control (temperatures, speed scaling, extrusion dynamics).

**Advanced Material Management:** Independent per-bay configuration (custom starting volumes, configurable residual safety buffers, and unrestricted spool swapping).

**True Production Persistence:** The machine retains its full operational capabilities seamlessly across power cycles, reboots, and multi-day production builds.

This is why serious manufacturing facilities, bureaus, and enterprise labs partner with us, and why they continue expanding their deployments with **PrintOverride** as their fleets grow. That is the stark difference between a fragile home-shop workaround and a bulletproof, commercial-grade operating environment engineered for real industrial hardware.

**For anyone looking for a reliable, production-proven deployment: reach out via our platform at [PrintOverride](https://printoverride.com) before October 1st and mention this GitHub discussion thread. We’ll provide a 20% discount on all our enterprise solutions, including our complete transformation suites for industrial platforms, up to converting legacy Fortus 900mc systems into fully unlocked, all-material F900 equivalents 😉**

dan1282, 2026-09-15:
We’ve had the printoveride solution for over a year now and are quite pleased with the performance and support from them. Material freedom is the biggest draw but also considering the upgraded print heads they offer now as well. Not relying on overpriced stratasys filament has allowed us to expand our fleet. Admittedly, we have supplemented the f370 with 2 bambu h2s printers but that’s a whole different topic lol.

fortususer, 2026-09-15:
“Re-flashing the controller” isn’t really what’s happening here. For context, the full OEM F123 firmware package is roughly 270 MB; ours is about 3.4 MB, and does not rewrite the printer’s firmware.

That’s a very different mechanism from the flash-wear and bricking scenario being described. Concerns raised so far have depended on assumptions that simply don’t match our implementation, and the failure mode seems to change each time that’s clarified. This will be my last reply on it.

wowgrant, 2026-09-15:
Leaving this final technical note not for the vendor above,who has made it clear they are bowing out,but for the engineers, lab managers, and shop owners evaluating these architectures.

Arguing Against Manufactured Strawmen:

No one ever claimed this workaround was "re-flashing a controller", that is an invented talking point. In fact, referring to "re-flashing the controller" on an F123 machine betrays an unfamiliarity with the hardware itself. Unlike other platforms like the Fortus 380mc/450mc that utilize dedicated controller boards, the F123 platform runs on a unified Single Board Computer (SBC) driving peripheral interface boards. It is a completely different architecture.

Fundamental Misunderstanding of the Operating System:

Claiming that the full OEM firmware is "roughly 270 MB" demonstrates a total lack of comprehension regarding the Stratasys software stack. The official 270 MB OEM UPG is merely an incremental delta payload containing select user-space binaries. The actual embedded operating system and underlying image partitions are gigabytes in size.

The 3.4 MB Reality:

Admitting the payload is only 3.4 MB confirms exactly what was pointed out earlier: this is an isolated, surface-level binary swap in mutable user-space. It cannot touch or integrate into verified system partitions without violating the hardware chain of trust, failing system core verity checks, and dropping the whole system into a black screen or unbootable state.

**Zero Capability for an Actual Ecosystem:**

Because this workaround cannot natively integrate into verified system space without failing integrity checks, it is architecturally dead-ended:

It cannot provide persistent remote telemetry or machine-side software access.

It cannot grant operators true control over print temperatures, speed profiles, or live extrusion parameters.

It cannot offer dynamic multi-bay spool accounting or residual safety buffers.

A 3.4 MB volatile hook that requires manual execution after reboots is a crude script injection, not an industrial platform upgrade. In production manufacturing, reliable hardware requires full system integrity and deterministic architecture, not ephemeral stopgaps.

DrewWing, 2026-09-15:
If anyone who's not selling "solutions" finds any more *real* info and is willing to open source it then please @ me. I'm unsubscribing from the ai slop advertising mess that has become this discussion.

Tomblarom, 2026-09-15:
For the sake of the original spirit of this repo:

> [!CAUTION]
> ### Stop advertising or debating any commercial solutions! Otherwise this issue will be closed!

wowgrant, 2026-09-15:
> For the sake of the original spirit of this repo:
> 
> Caution
> ### Stop advertising or debating any commercial solutions! Otherwise this issue will be closed!

100% agreed. Closing this thread is actually the best outcome.

The technical realities, and hardware risks have been thoroughly laid out for anyone looking for industrial reliability. Everything else at this point is just noise.

**Lock it down and let people evaluate the technical facts for themselves.**

dan1282, 2026-09-15:
Here’s my take: I’ve been looking for a solution to affordable filament for the f370 since we first purchased it when it came out in 2016 or 2017. It became very apparent that the eeprom programmers for the dimension printers was not going to happen anytime soon. Then 2-3 years ago a company started selling reprogrammed chips with their filament exclusively but required you to plug into the usb port and have an external feed. Never ended up trying it. Now there are at least 2 commercial solutions (1 which we adopted). It works seamlessly with the printer and easily paid for itself within 10 spools of material. Sure it wasn’t free but I’m also not using the f370 for personal use. It’s heavily used at work for prototyping. 

The bigger thing I see is that in the years since this has come out, the consumer grade printers have advanced to a point that it would be hard for our use case to even justify ever buying another stratasys. Bambu really has an impressive line of printers that arguably outperform the f370 in a lot of areas. Our last maintenance visit before we cancelled our contract, the tech was telling us that he’s seeing bambu being implemented in a lot of his sites and the frequent comment is “it’s good enough.” Sure bambu requires a lot more tinkering and you may not get the exact precision of a stratasys but they are significantly faster, cheaper, and can run any material stratasys can and more out of the box. And if it breaks, the absolute worst case scenario is I can buy a new one and still be ahead of the game.