bvanheu/stratatools #85: Infos from eeprom printing

Gegounet, 2022-05-18:
Hi,
I don't know if somebody is still here but i will try anyway... Fingers crossed
So when i try the stratatools eeprom_decode that error comes out ;
stratatools eeprom_decode --machine-type fox --eeprom-uid b3b6471dd014103f originalb3-1014d01d47b6.bin
Traceback (most recent call last):
  File "/usr/local/bin/stratatools", line 11, in <module>
    load_entry_point('stratatools==3.0', 'console_scripts', 'stratatools')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "build/bdist.linux-armv7l/egg/stratatools/console_app.py", line 13, in <module>
  File "/usr/local/lib/python2.7/dist-packages/protobuf-4.21.0rc1-py2.7.egg/google/protobuf/text_format.py", line 52, in <module>
    from google.protobuf.internal import decoder
  File "/usr/local/lib/python2.7/dist-packages/protobuf-4.21.0rc1-py2.7.egg/google/protobuf/internal/decoder.py", line 86, in <module>
    from google.protobuf.internal import containers
  File "/usr/local/lib/python2.7/dist-packages/protobuf-4.21.0rc1-py2.7.egg/google/protobuf/internal/containers.py", line 73
    def __init__(self, message_listener:any) -> None:
                                       ^
SyntaxError: invalid syntax

Any ideas ?
Thanks :)

eklmv, 2022-08-14:
Latest version of protobuf doesn't support python2, i tried to build it with protobuf=3.12.0 and it works.

Gegounet, 2022-08-24:
Yes thank you It worked :)